bambulab/BambuStudio #4101: Color by Height range colors deeper layers too

UBaer21, 2024-05-18:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Win10

### Printer

P1P AMS

### How to reproduce

color something complex per height range in different levels

### Actual results

[image]
[image]

second example with thicker magenta stripe
[image]
[image]

### Expected results

If i color layer 3-8 i want color only in layer 3-8  and not in other layers causing additional filament changes.
It would also help if i can color actual layers and not any height that leads to weird results

### Project file & Debug log uploads

[24-147-Prawned.bambu.3mf.zip](https://github.com/bambulab/BambuStudio/files/15360939/24-147-Prawned.bambu.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-05-20:
Due to the limitations of the current coloring algorithm, the boundary of height_range is not a true straight line. This is an item to be optimized in the future.

MadMax389, 2024-05-21:
You can get well-defined color layers by using a height range modifier instead of painting them. (You just can't see the colors in the prepare screen).
[image]

UBaer21, 2024-05-22:
@MadMax389  I'll use that in future - good advice, thanks for the tip,  Brings us to the next issue - why is this not displayed in prepare?