bambulab/BambuStudio #4103: Outerwall speed of the bridging layer is different from the previous and next layers on small model.

raphael008, 2024-05-18:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10 (19045.4412)

### Printer

Bambu Lab P1S

### How to reproduce

1. Filament: Generic PETG. Process: 0.20mm Standard.
2. Adding a **20mmx20mmx20mm** cube.
3. Slicing it and jump to layer 96. 
4. Switching the Color Scheme to Speed and you can see the outerwall speed is near 159mm/s. The speed of layer 95 is about 20mm/s. Looking into these 2 layers. The outerwall speed differ so much. This will reduce the printing quality of outerwall.

### Actual results

Slicing result:
[image]

Outerwall speed of layer 96:
[image]

Actual printed result:
[image]

### Expected results

1. Filament: Generic PETG. Process: 0.20mm Standard.
2. Adding a 80mmx80mmx80mm cube.
3. Slicing it and jump to layer 96. 
4. Switching the Color Scheme to Speed and you can see the outerwall speed is same as layer 95 and layer 97.
5. Printing the cube and the outerwall issue should be addressed and look great.

Slicing result:
[image]

### Project file & Debug log uploads

This issue is related to slicing algorithms, no log files or project file needed.

### Checklist of files to include

- [X] Log file
- [X] Project file

raphael008, 2024-05-18:
中文版本：

问题表现：在小模型上(20mmx20mmx20mm)，顶面桥接层与上一层（稀疏填充层）的外墙速度变化过大，导致实际打印物体外表面有肉眼可见的收缩以及断层感。

个人调查：
1. 分别将模型调整到以下三种规格(A. 20mm x 20mm x 20mm), (B. 80mm x 80mm x 20mm), (C. 80mm x 80mm x 80mm)。
仅在A规格中发现以上问题，此处可以推断问题发生的原因和X/Y尺寸有关系。
2. 调整层时间为0后再次切片，问题解决，外墙速度均一致，此处推断问题发生的原因和层时间有关系。

问题总结：
切片算法在处理小模型时会由于层时间的原因，急剧加快顶面桥接层的外墙速度，导致实际打印件产生外观问题。

QingZhangBambu, 2024-05-20:
你说的没错，这是层时间不同引发的外观问题。是个值得改进的点。我会把它加入feature list, 并考虑更好的解决方案。

raphael008, 2024-07-25:
Hi @QingZhangBambu https://github.com/SoftFever/OrcaSlicer/pull/5148 orca那边有人做了这个功能，能合并过来吗？

QingZhangBambu, 2024-07-26:
@raphael008 您好，我们已经另外实现了用参数来控制不同悬垂度的走线是否参加冷却降速的功能。只是需要下次正式版本发布才能您才能获取。

raphael008, 2024-07-26:
Hi @QingZhangBambu 
谢谢，能看到这个有所改变真的是太好了👍

closed 2024-07-26 by raphael008