bambulab/BambuStudio #4105: Chamber fan at 100% while leveling

Kimonokosch, 2024-05-18:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1S

### How to reproduce

Perform bed leveling or filament purge.

### Actual results

Chamber fan speed goes to 100% when bed leveling or filament purge starts. (also when standard leveling before print runs)

### Expected results

Chamber fan should stay off or low before printing.

### Project file & Debug log uploads

No file from today there.
Problem appeared directly after updating Bambu Studio.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ValeriaMomo, 2024-05-18:
i have the same issue

DanTup, 2024-05-19:
There's an existing issue about this at https://github.com/bambulab/BambuStudio/issues/4054

I was prompted with a "configuration update" that has reverted that change and it no longer does this.

pmoliterni, 2024-05-19:
@DanTup I was not prompted with any "configuration update". Is there a specific procedure to follow ? Thank you.

DanTup, 2024-05-19:
Unfortunately I don't know... I was prompted with another app update (from .66 -> .67) and after installing it I started another print (but did not get the config update), and then when I opened another project, that's when it popped up saying there was a config update.

You could try clicking Help -> Check for Updates but I don't know if the configuration updates are checked there or only the app (nor do I know if the .67 update was required before the configuration update appeared).

pmoliterni, 2024-05-19:
@DanTup I've tried. It says that I have the last version 1.9.66.

SaltWei, 2024-05-20:
> @DanTup I've tried. It says that I have the last version 1.9.66.

Please pay attention to the right bottom corner of studio after it being launched.
[image]

pmoliterni, 2024-05-20:
It appeared, thank you.

SaltWei, 2024-05-21:
> It appeared, thank you.

You are welcome :)

closed 2024-05-21 by SaltWei