bambulab/BambuStudio #4108: Max. Bed Temperature is FORCED upon Print Start on Newest Firmwares (P1P)

bolsoncerrado, 2024-05-19:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Sonoma 14.4.1

### Additional system information

M1Ultra

### Printer

P1P

### How to reproduce

1. Define Machine GCode Preheating scheme to avoid UPS overload
2. Define Max. Bed temp for filament to 55, 60 or whatever
3. Launch a print with that filament and machine selected

### Actual results

Upon launching the print, loading the gcode on the printer and "starting" to read the .gcode, printer sets the bed preheat temp to its maximum automatically. THEN after it has been reached, it starts reading the MACHINE DEFINED .gcode

### Expected results

To just read the machine defined gcode in regards to bed temp gcodes.

### Project file & Debug log uploads

The explanation is simple: 

I run a farm of MANY P1Ps. 
Every 2 P1Ps are hooked to a UPS supply.
IF one or more printjobs are sent to P1Ps with the latest firmware on the same UPS, the printers try to PREHEAT the bed to its maximum first, THEN read the gcode. 
On previous firmwares this wasn't happening, the printer just read the gcode and used MY altered gcode to preheat the bed SLOWLY so the UPSs don't fail with an OVERLOAD alarm that requires me to phyiscally reset each UPS.

This is the GCODE I was using and that now is read AFTER the max heatbed temp is reached, which renders it USELESS because the OVERLOAD on the UPS are triggered :'(

`
M140 S10
M190 S10
M140 S15
M190 S15
M140 S20
M190 S20
M140 S25
M190 S25
M140 S30
M190 S30
M140 S35
M190 S35
M140 S40
M190 S40
M140 S45
M190 S45
M140 S50
M190 S50

M140 S[bed_temperature_initial_layer_single] ;set bed temp
M190 S[bed_temperature_initial_layer_single] ;wait for bed temp
`

(I tried to use your logical operators IF THEN ELSE but they were not working fine for me for whatever reason).

I hope this makes any sense and I still wonder WHY did you introduced such change of forcing the max bed temp and THEN read the machine gcode rendering it useless. 

I'm open to your questions!

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SaltWei, 2024-05-20:
Seems to be firmware issue？

hechuan949494, 2024-05-20:
Please upload log through Handy APP,and give us your ticket number

bolsoncerrado, 2024-05-20:
I don't have any log to upload, all the p1ps forcedly run 1. 4.x due to
this issue.

bolsoncerrado, 2024-05-20:
Just for the record, I don't have any log to upload, all the p1ps forcedly run 1. 4.x due to this issue.

fastiuk, 2024-07-04:
@bolsoncerrado I have the same issue with my A1.
And I found a workaround. In the filament setting set "Initial layer" temperature to 20.
Then in machine g-code you need to do like that:

```
{if bed_temperature[current_extruder] > 30}
M140 S30
G4 S20
{endif}
{if bed_temperature[current_extruder] > 40}
M140 S40
G4 S20
{endif}
{if bed_temperature[current_extruder] > 50}
M140 S50
G4 S20
{endif}
{if bed_temperature[current_extruder] > 60}
M140 S60
G4 S20
{endif}
{if bed_temperature[current_extruder] > 70}
M140 S70
G4 S20
{endif}
{if bed_temperature[current_extruder] > 80}
M140 S80
G4 S20
{endif}
{if bed_temperature[current_extruder] > 90}
M140 S90
G4 S20
{endif}

M140 S{bed_temperature[current_extruder]}
```

And the last step is to find `M190 S{bed_temperature_initial_layer_single}; ensure bed temp` and replace with `M190 S{bed_temperature[current_extruder]}; ensure bed temp`

If you need to have some difference between the initial and other layers bed temperature you can apply it by adding/substracting it here in g-code.

gaolegao-lx, 2024-07-05:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

bolsoncerrado, 2024-07-05:
> @bolsoncerrado I have the same issue with my A1. And I found a workaround. In the filament setting set "Initial layer" temperature to 20. Then in machine g-code you need to do like that:
...
> This workaround will use very low initial temperature to trick a printer to skip initial preheat, but instead will use temperature of other layers to set required temperature.

Many thanks! WIll try your solution and see if it's a nice workaround on the P1 series too ;)