bambulab/BambuStudio #4109: Bambu Studio 1.9.1 crashes upon opening on windows

Parsentaur, 2024-05-19:
### Bambu Studio Version

1.9.1

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Microsoft Windows 10 Pro, 10.0.19045 Build 19045

### Additional system information

CPU Ryzen 7 5800X3D
GPU NVIDIA GeForce RTX 4070
RAM 64gb 3600MHz DDR4

### Printer

Bambu P1s

### How to reproduce

Opened Bambu Studio, loaded for 5-10 seconds, Bambu studio crashed 

### Actual results

App crashes

### Expected results

App does not crash

### Project file & Debug log uploads

Unable to find debug logs

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2024-05-22:
@Parsentaur 
Please provide Studio logs with DMP files.
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

augdevelo, 2024-05-22:
Same issue happens. Here is what's logged in eventvwr:
[image]

Installed latest Windows 10 SDK with same result.