bambulab/BambuStudio #4116: Supports with PLA_SUPPORT interface are very difficult to remove

Gerard26260, 2024-05-19:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambulab X1C

### How to reproduce

Print a file with support using PLA_SUPPORT at the interface

### Actual results

The main support is easy to remove but at the interface with the printed object it is impossible to remove the pla-support without scraping. It was easier with version 1.8.

### Expected results

The interface support must go directly with the interface
[image]
 body

### Project file & Debug log uploads

[Beagle.zip](https://github.com/bambulab/BambuStudio/files/15370172/Beagle.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Gerard26260, 2024-05-23:
In the flush volume window, the multiplier parameter was set to 0.
I have just put it to 0.6 and it seems to be better.
This can explains the mixing of PLA and PLA-SUP, complicating the interface and making it difficult to separate the support from the object.
I will confirm you if everything is ok .

tangketan, 2024-05-24:
@Gerard26260 Thanks for your clarification. It seems it's another bug (default multiplier parameter should be 1). I'll report to other colleagues.

tangketan, 2024-05-24:
@Gerard26260 By the way, how do you get the 0 multiplier? It's non-zero in the 3mf, and I've printed it with pretty good result.
[image]

ClawzSkunk, 2024-08-26:
When is this being fixed?