bambulab/BambuStudio #4117: Supports with PLA_SUPPORT interface are very difficult to remove

Gerard26260, 2024-05-19:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambulab X1C

### How to reproduce

Print a file with Bambula PLA basic filament and support using PLA_SUPPORT at the interface

### Actual results

The main support is easy to remove but at the interface with the printed object it is impossible to remove the pla-support without scraping. It was easier with version 1.8.

### Expected results

The interface support must go directly with the main support

### Project file & Debug log uploads

[Beagle.zip](https://github.com/bambulab/BambuStudio/files/15370172/Beagle.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Gerard26260, 2024-05-19:
This issue is also reported by other users on the reddit server

Gerard26260, 2024-05-19:
Issue has been closed by error

zhimin-zeng-bambulab, 2024-05-20:
https://github.com/bambulab/BambuStudio/issues/4116 Duplicate BUG, ​​only keep one

closed 2024-05-20 by zhimin-zeng-bambulab

doncaruana, 2024-07-27:
@zhimin-zeng-bambulab - This bug is not fixed. I tested the exact same model with 1.9.03.50 and 1.8.04.51 with the same model and the supports are relatively easy to remove with 1.8 and almost impossible with 1.9.