bambulab/BambuStudio #4120: Dozens of people agree Supports are harder to remove and partially welt to parts.

SxeyChick, 2024-05-20:
### Bambu Studio Version

1.9.1

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1S

### How to reproduce

Every single print will have worse support performance. 

### Actual results

https://github.com/bambulab/BambuStudio/assets/115029650/0f167a45-ab09-46de-a7ed-ec4337425e8f

### Expected results

An improvement between release, not a downgrade. 

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MartinChan3, 2024-05-21:
@SxeyChick Do you use tree-support(default setting) and variable-layer-height function at the same time ? Try closing variable-layer-height, I get a better support to remove ,and no welt parts left on the model.

SxeyChick, 2024-05-22:
> @SxeyChick Do you use tree-support(default setting) and variable-layer-height function at the same time ? Try closing variable-layer-height, I get a better support to remove ,and no welt parts left on the model.
> 
> 

Yes I'm using tree. And "variable layer height" isn't an option in BambuSlicer. If you're referring to another option, please refer to it by the name it is called In the slicer options.

capsel22, 2024-05-22:
> Yes I'm using tree. And "variable layer height" isn't an option in BambuSlicer. If you're referring to another option, please refer to it by the name it is called In the slicer options.

Bambu Studio does have varable layer height 
[image]

SxeyChick, 2024-05-23:
> Bambu Studio does have varable layer height
>
> [image]

That one? You're recommending to "close" it, but it's off by default. So that's definitely not what's causing issues for me and the many others.

Antmax, 2024-05-24:
Same problem here and several people complaining on reddit. It seems the interfacing part of the support completely fuses to the model. The supports are much more sturdy, but they just done come off cleanly like they used to.

Antmax, 2024-05-24:
[image]

Here's a pic comparing latest 1.9.x and 1.8.4

I printed the orc twice with the same result as 1.9. Today I just printed the soles of the feet in 1.8 to compare in case I was imagining things. As you can see in the 1.8 pic, the supports pull off fairly cleanly while 1.9 was fused. 

What you see in the 1.9 side is after using needle pliers, side cutters and a modelers file. 1.8, I just grabbed an edge of the support with my pliers and stared to peel it off like a stubborn orange skin.

The flat bed plate textured part of the foot being tricky is expected because it was making direct contact with the plate. But on the 1.9 everything is fused so badly I even pulled a hole in the heel trying to get the support off. 

It does it with every model. You can see it on the underside of the top half of the orc in the 1.9 pic too.

SxeyChick, 2024-05-28:
I've been avoiding printing things with supports since this update came out. But my next part needs them. Has this been fixed yet?

tangketan, 2024-06-14:
The latest version (1.9.2) has improved on this issue. Please take a try. Or if you have the 3mf files, please share them so I can double check if the problem has gone.

Malice890, 2024-06-14:
> The latest version (1.9.2) has improved on this issue. Please take a try. Or if you have the 3mf files, please share them so I can double check if the problem has gone.

For me 1.9.2 tree supports were eaiser to remove, but still slightly fused to the print. 1.9.1 they were fully fused. 1.8.4.51 they work perfect.

SxeyChick, 2024-06-15:
> The latest version (1.9.2) has improved on this issue. Please take a try. Or if you have the 3mf files, please share them so I can double check if the problem has gone.

I agree with the other commenter. It's better than 1.9.1, but worse than it was before. 

I am STILL putting projects on hold till I can use supports again. 

I can actually help you guys fix it. Here:

Copy support code from before 1.9.1, paste in the next update, releases it tomorrow. 

Pretty simple.

Malice890, 2024-06-15:
> I agree with the other commenter. It's better than 1.9.1, but worse than it was before.
>
> I am STILL putting projects on hold till I can use supports again.
>
> I can actually help you guys fix it. [...]

I rolled back to 1.8.4.51. Will try again in 1.9.3 if they fix the fusing issue.

Malice890, 2024-06-21:
@tangketan Anymore updates on this issue?

MartinChan3, 2024-07-30:
@Malice890 I have tried the newest release, and the tree support has been easier to be removed

ClawzSkunk, 2024-08-26:
I'm still getting a lot of issues with the supports still fusing to the prints.. This is getting ridiculous

Malice890, 2024-08-26:
I tried the newest release again to make sure it was not just me, and broke 2 prints trying to get the fused tree supports off. Rolled back to 1.8.4.51 and they come right off.

MartinChan3, 2024-08-26:
I remember the devoloper has explained the reason why some supports are still 'not easy to be removed' due to it's a balance between 'easy to be removed' and 'good overhang', it seems that the good removal characteristic was an 'lucky error' for V1.8.4.
I have to say maybe some wiser method could be made to solve this problem(especially when developer has realized it is not a technical problem but a product concept problem), have you use the newest Orca?May be a good competitor performance could make them do more...

rapidcloubr, 2024-09-02:
I was having the same problem, I reverted to version 1.8.4 and everything started working again.

MartinChan3, 2024-10-31:
I believe each one who suffered this issue will be reliefed in 1.10 Beta2 and this beta version's support is so easy to be removed!
https://bambulab.feishu.cn/sheets/RsbfsQxSBhmO8KtM1N0cZZxNnWf?from=from_copylink

Malice890, 2024-10-31:
> I believe each one who suffered this issue will be reliefed in 1.10 Beta2 and this beta version's support is so easy to be removed! https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.81

I hope you are right, I will wait for the non beta and try it again.  I do not want to live on 1.8.4.51 forever since the supports come off easy.

cybrdyne, 2024-11-15:
Same problem here, using the latest version as of 11/15/2024, and all supports are fused with a top/bottom z of 0.2mm. Rolling back to 1.8.4 resolves this.

Malice890, 2025-01-11:
> 1.8.4.51

Really depends on the model, BS 1.8.4.51 the supports were perfect.. the current version it is hit and miss for me. Some things work great, others fuse and I have to bust out the torch and pretty it up the best I can.