bambulab/BambuStudio #4122: Negative Thickness Get App Crashing

hechay, 2024-05-20:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab P1S

### How to reproduce

1. Click on Object
2. Select the Object
3. Choose the Text Shape button
4. Choose the negative thickness
5. Get result by app crashing

### Actual results

App will be crashing when I choose the negative text thickness ( like -1, -2.. etc )

### Expected results

We should check the negative value.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/15374830/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

hechay, 2024-06-03:
Hope this bug will be fixed in next version. Just an if-else when get input the text field. Thanks.

zhimin-zeng-bambulab, 2024-06-04:
It will be fixed in next version

zhimin-zeng-bambulab, 2024-06-12:
@hechay  It has been fixed in v01.09.02.57, please have a try

closed 2024-06-12 by zhimin-zeng-bambulab