bambulab/BambuStudio #4124: Elephant Foot Compensation Conflicts with Initial Layer Line Width

Xiaoxun92, 2024-05-20:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 14.5

### Printer

A1 Mini

### How to reproduce

1. Import the attached Elephant Foot Bug Test.stl file. Note that the initial layer is very thin.
2. Set `Line width -> Initial layer` to 0.5mm, and set `Elephant foot compensation` to 0.3.
3. Slice the model. **Observe the diameter of the initial layer's outer wall.**
4. Reduce `Line width -> Initial layer` to 0.3mm, and set `Elephant foot compensation` back to 0.
5. Slice the model to refresh things.
6. With `Line width -> Initial layer` kept at 0.3mm, set `Elephant foot compensation` to 0.3mm again.
7. Slice the model. **Observe the diameter of the initial layer's outer wall.**
8. Set `Line width -> Initial layer` back to 0.5mm, while keeping `Elephant foot compensation` at 0.3mm.
9. Slice the model. **Observe the diameter of the initial layer's outer wall.**

### Actual results

In step 3, the diameter of the initial layer didn't change in response to the 0.3mm `Elephant foot compensation` value.

In step 7, the `Elephant foot compensation` takes effect with a thinner `Line width -> Initial layer`.

in step 9, `Elephant foot compensation` still takes effect despite the `Line width -> Initial layer` has returned to the initial value.

### Expected results

In step 3, if `Elephant foot compensation` is ignored when a wall is very thin, the user should ideally get a warning message about that.

In step 7, the outcome looks right.

in step 9, `Elephant foot compensation` should behave the same as in step 3. The discrepancy indicates a possible conflict bug between `Elephant foot compensation` and `Line width -> Initial layer`. 

### Project file & Debug log uploads

[Elephant Foot Bug Test.zip](https://github.com/bambulab/BambuStudio/files/15380177/Elephant.Foot.Bug.Test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2024-05-21:
there is the slice step: 
step 1: apply elephant foot compensation first, get the shrinked expolygon,
Step 2: generate the wall with wall width. 
if the elephant compensation change, do step 1 & 2, so you get result in your step 
if only change wall width, just do step 2. so you get the result in your step 9
This is a bug that needs to be improved, but it involves the issue of how to use memory data, so it is difficult and needs to be discussed carefully.

Xiaoxun92, 2024-05-21:
@QingZhangBambu Thank you for your comment!

From a user's perspective, I see two issues:
1. The elephant foot compensation setting can sometimes be silently ignored (as seen in step 3), which might catch the user by surprise. This is how I discovered this issue :joy:. I suggest raising a notification to the user when this happens (I assume the compensation is ignored intentionally in this case?)
2. With the same settings, the slicer generates different results (step 3 vs. step 9), which is clearly undesired. I assume this is the bug you were referring to.

QingZhangBambu, 2024-05-22:
for issue 1:  
if the outer polygon is too small, the elephant's foot will not be applyed. This should be written on the tooltip and will be improved later

Xiaoxun92, 2024-05-22:
@QingZhangBambu Thank you! Please feel free to close this ticket when appropriate.