bambulab/BambuStudio #4125: After last update bambustudio://open?file= stop to work on windows

abisewski, 2024-05-20:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Built from source by yourself or others

### OS version

Windows 11

### Printer

Any Bambu Lab

### How to reproduce

Paste this link on your browser:

bambustudio://open?file=https://static.stlflix.com/Keg_o_Beer_Can_Holder_Bambu_Lab_b750eedde5.3mf

### Actual results

Just open Bambustudio

### Expected results

Open Bambustudio, download and open the project.

On MacOS is working fine (with MacOS link: bambustudioopen://https://static.stlflix.com/Keg_o_Beer_Can_Holder_Bambu_Lab_b750eedde5.3mf)

### Project file & Debug log uploads

https://static.stlflix.com/Keg_o_Beer_Can_Holder_Bambu_Lab_b750eedde5.3mf

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2024-05-21:
@abisewski 
We do not support loading 3mf files in this way because it is very unsafe. Trojan software or viruses can be downloaded to computers in this way and cause harm.

abisewski, 2024-05-21:
@walterwongbbl Is there any way to use this method to directly open/download files in Bamboo Studio, or is it just no longer supported?

xperiments, 2024-06-19:
@walterwongbbl TBH I don't see the point...

Why windows yes and mac no?

Also the "security" you added take me only 1m to bypass, is quite weak

https://github.com/bambulab/BambuStudio/blob/d19c9654783701d792bdf4c8fc3480d91ea991d3/src/slic3r/GUI/GUI_App.cpp#L1093