bambulab/BambuStudio #4126: K Factor Missing/Broken on Device Tab

SpannMagoo, 2024-05-20:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

Device name	SeanR
Processor	13th Gen Intel(R) Core(TM) i9-13950HX   2.20 GHz
Installed RAM	32.0 GB (31.7 GB usable)

System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

Edition	Windows 11 Home
Version	23H2

### Printer

Bambu Lab P1S w/AMS

### How to reproduce

Go to Device Tab in the latest Bambu Slicer and the K Factor is not shown and can't be edited.

[image]

[image]

### Actual results

Nothing Shows

### Expected results

To see and edit the k factor

### Project file & Debug log uploads

Why is this needed when its Bambu Studio? It wont let me attached anyways.

### Checklist of files to include

- [ ] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-05-21:
@SpannMagoo  The new version of firmware can support storing k values ​​on the machine like X1C. If you want to create your own calibrated values, you can add them in the management results of the calibration page.
[image]

SpannMagoo, 2024-05-21:
I am on a P1S and I haven't the slightest clue how to do that and Orca shows it just fine.

MadMax389, 2024-05-21:
You have to run a manual calibration of your filament which will generate a calibration profile. You can see and edit that profile on the Manage Result Screen.
[image]