bambulab/BambuStudio #4132: 1.9.1.67 - Showstop: Layer Pause Bug Gcode on the wrong layer.

TinyMito, 2024-05-21:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab github releases

### OS version

macOS

### Additional system information

MacBook Pro 14

### Printer

P1S Combo

### How to reproduce

I found out the gcode is pausing at 2 layers offset.

1. Add pause to any layers ie. **Layer 57** in the screenshot.
2. Printer actually delayed paused by two layers.
3. I have gone in to investigate the gcode.
4. It actually paused at **layer 59** instead.

### Actual results

[image]

### Expected results

Should pause on the correct layer.

### Project file & Debug log uploads

not available.

### Checklist of files to include

- [ ] Log file
- [X] Project file

TinyMito, 2024-05-21:
For 0.16mm layer height, same problem but now it is offset by 3 layers.

[image]

TinyMito, 2024-05-21:
Additional updates, it seems like it only affect model with a base raft or with support. I can provide the project file in private only. Please let me know where to share it.

LiZ-Li-BBL, 2024-05-31:
Hi,
You could share the file to liz.li@bambulab.com, we will check the problem.

moop2000, 2024-06-02:
I am experiencing the same issue. I've narrowed it down to the scarf joint seam feature being enabled. It doesn't matter if it's enabled per model or globally, once it is enabled in the file, the pauses will be offset. The number of layers in error appear to be affected by the layer height. In my testing, .16 layer height causes the pause to be adjusted by three layers, while .2 layer height is adjusted by 2 layers.

[Layer Bug.zip](https://github.com/user-attachments/files/15524162/Layer.Bug.zip)
I've attached an example file. It is a standard setup with all default settings. Plate 1 is just a cube, with a pause at 10mm, which is offset. Plate 2 has a cube with scarf joint enabled just for that one object, not globally, and it is also offset. Removing that second object, plate 1 will show the proper pause location.

nigeljohnson73, 2024-06-02:
I have just bumped into this. Also narrowed it down to scarf joins (as this is the only thing I added in my project) I'm using 0.2mm layer height, and using 20mm scarf length, the offset for me is 4 layers - requested start of layer 14
[image]

and it happens the start of layer 18
[image]

How is that even linked :D

nigeljohnson73, 2024-06-02:
Oh, interesting proof, and if you need scarf seams... Set the pause at the place with the standard profile. Then enable scarf seems with the settings you need, and the pause will "move", but will still occur at the right place in printing.

TinyMito, 2024-06-03:
> Hi, You could share the file to [liz.li@bambulab.com](mailto:liz.li@bambulab.com), we will check the problem.

The file, I sent, has scarf seam enabled. Maybe that's why the issue like many others are facing the same problem.

QingZhangBambu, 2024-06-03:
i will fix it as fast as possible

TinyMito, 2024-06-11:
[image]

Looks good and thank you for the fix!