bambulab/BambuStudio #4134: 1.9.1.67 - Force saving .gcode.3mf does NOT work always on MacOS Somona 14.x

bolsoncerrado, 2024-05-22:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Sonoma 14.4.1

### Additional system information

M1Ultra

### Printer

X1C

### How to reproduce

1. Work on a 3mf project
2. Slice
3. Export gcode
4. Click on existing name on the file name list (no extension is assigned  yet, CORRECT!)

### Actual results

5. System reports the .3mf is gonna be overwritten!

### Expected results

5. Saves as .gcode.3mf instead!

### Project file & Debug log uploads

https://github.com/bambulab/BambuStudio/issues/3375

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bolsoncerrado, 2024-05-22:
[image]

bambu123, 2024-05-24:
It's not a very serious problem. Saving as .3mf doesn't affect reopening. It will be improved in subsequent versions.

bolsoncerrado, 2024-05-24:
The "non serious problem" is the fact you overwrite the project file with just the gcode file, ruining the source of the whole project... :(

bolsoncerrado, 2024-08-30:
You sneakedly patched this in 1.9.4 but it is still buggy at least in MacOS version.

When "exporting plate gcode", by default the system adds .gcode.3mf:

[image]

If you click OK to save as is, it is added TWICE!

[image]

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00

bolsoncerrado, 2024-10-17:
@Haidiye00 apparently you've now removed the "Overwrite warning" when clicking an existing file to save it with the same name. This was not the behaviour of past versions and can render to unwanted overwrites. Perhaps making it a user selectable setting?

Haidiye00, 2024-10-17:
Sorry, if you want to save another copy, please modify the file name

bolsoncerrado, 2024-10-17:
LOL but you can't fix a bug and generate another! What the....!?