bambulab/BambuStudio #4135: printer profile does not remove old printer

sebastienkoopman, 2024-05-22:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10

### Additional system information

Not applicable

### Printer

bambu lab x1 carbon

### How to reproduce

bambu studio open
 tab prepare.
drop down 2 printers
printer settings remove all
select bambu x1c  4mm selected.
confirm
and still 2 printers in drop down.

### Actual results

too many printers in drop down than in selected settings menu.

[image]
[image]

### Expected results

1 printer selected but 2 in drop down also second is primary need a fix.

### Project file & Debug log uploads

Not applicable

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-05-24:
[image]
I can not reproduce it here
could you give a gif and upload the log here?

ajurna, 2024-11-22:
https://github.com/user-attachments/assets/f107152b-5dc6-4e1d-ba5c-c89706981f12

i also have this issue also and here is a vid of it in action.

ajurna, 2024-11-22:
after restarting bambu studio it seems to have removed them.