bambulab/BambuStudio #4138: Paint issue

seanmcguire413, 2024-05-22:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Intel Core i5 7th gen

### Printer

X1C + 2 AMS

### How to reproduce

https://makerworld.com/en/models/221464
and slice using x1c I am using a profile with long retraction saved.
when on the preview page the ' in It's is all white but when it is sliced the ' is mostly brown from the base filament.  This wall is impossible to fix all painting options will not fix this.

### Actual results

the ' is brown and should be white. when on the preview tab and when printing.  it isn't white on the screenshot from makerworld so it might be a new bug.

### Expected results

the ' should not be brown

### Project file & Debug log uploads

[new+zelda+art+v24.zip](https://github.com/bambulab/BambuStudio/files/15406517/new%2Bzelda%2Bart%2Bv24.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-05-28:
@seanmcguire413  Could you describe your problem in more detail with screenshots?

seanmcguire413, 2024-05-29:
@zhimin-zeng-bambulab 
prepare tab the ' is all white
[image]
after slicing the ' is brown because of the default color of the design.
[image]

seanmcguire413, 2024-05-29:
no matter what I do to pain the ' it will always be brown

zhimin-zeng-bambulab, 2024-05-30:
@seanmcguire413  Thank you for your patient explanation!

zhimin-zeng-bambulab, 2024-06-12:
@seanmcguire413  It has been fixed in v01.09.02.57, please have a try

closed 2024-06-12 by zhimin-zeng-bambulab

seanmcguire413, 2024-06-12:
Confirmed fixed! Thank you so much!