bambulab/BambuStudio #4139: Inner/outer/inner wall order setting is sometimes ignored

ericdlaspe, 2024-05-22:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma 14.4.1

### Printer

Bambu Lab X1 Carbon

### How to reproduce

[Peg 1 v37.step.zip](https://github.com/bambulab/BambuStudio/files/15406398/Peg.1.v37.step.zip)

1. Add attached STEP file to plate (or open the attached project)
2. Split the model to objects delete "base" and auto-orient "core" to lay it on the long, flat face
3. Change "Order of walls" to "inner wall/outer wall/inner wall"
4. Change "Wall loops" to 3
5. Slice/preview
6. Navigate to layer 35 or 115 (for example) and notice that inner walls are both printed before outer wall

### Actual results

[image]
Notice in the screenshot that both inner walls have printed before the outer wall

### Expected results

The innermost wall should be printed, followed by the outer wall, followed by the middle wall.
This issue does not occur using the same model and settings in Orca Slicer 2.0.0.

### Project file & Debug log uploads

[Logs.zip](https://github.com/bambulab/BambuStudio/files/15406612/Logs.zip)
[Bambu Bug Repro 1.3mf.zip](https://github.com/bambulab/BambuStudio/files/15406595/Bambu.Bug.Repro.1.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ericdlaspe, 2024-05-22:
This commit from Orca might be relevant. I'm not familiar with this code base, so I'm not sure. https://github.com/SoftFever/OrcaSlicer/commit/b736a0bc385217fb3440f4b9bf802de96dd51e90

QingZhangBambu, 2024-05-31:
only change the sequence with the neareast inner wall. i try to fix it.

QingZhangBambu, 2024-06-28:
it has been fixed and will be released in the next public version.

ericdlaspe, 2024-06-28:
Thank you! I would be interested in viewing the commit(s) if you don’t mind sharing. :)

megalanthus, 2025-03-13:
I noticed that the setting does not affect if the wall consists of three layers (the fill is always printed last), inner wall width = 0.45:
[image]
But if you reduce the filling thickness, the setting starts to work as it should, inner wall width = 0.3:
[image]

TheSlider2, 2025-08-13:
[test-Assembly.zip](https://github.com/user-attachments/files/21761300/test-Assembly.zip)

[image]