bambulab/BambuStudio #4141: Change behavior of auto-refill to account for filament stuck on spool

abomb60, 2024-05-22:
**Is your feature request related to a problem? Please describe.**
Some spools have a hole for the beginning of the filament and sometimes the AMS cannot pull the filament out of the hole; also it appears that Bambu has taped the start of the filament to some of their cardboard spool cores so this will cause the AMS to not load the end of the filament and causing an extruder stuck error (Extruding filament failed.  The extruder may be clogged. [0700 8007 113622]

**Describe the solution you'd like**
Enhance the logic so that when auto-refill is enabled to automatically switch to the next like filament in the AMS.  I'd rather have a few meters of filament left on a spool than a failed 48 hour print.

**Describe alternatives you've considered**
Only alternative is to babysit the printer/AMS and snip the end of the filament or manually remove the tape from the Bambu carboard spool core before the error presents.  If the user does this then the AMS does switch over to the next spool when auto-refill is enabled and configured correctly.

fhscholl, 2024-06-07:
> Bambu has taped the start of the filament to some of their cardboard spool cores

This is my issue as well. The Auto Refill feature is great, but it's pretty useless if I come back to a chunk of tape-wrapped filament stuck in the AMS, which has now happened more than once.

kilrah, 2024-08-29:
Same, when a spool gets tangled whether at the end because of a bend getting stuck in a hole or in the middle of a spool the auto-refill feature should switch to the next like spool automatically. If there is none we should have an option to change to any of the other AMS spools manually e.g. for prints that will be painted anyway and thos color doesn't matter.

ados8, 2025-05-25:
Year and we don't have a fix, this flaw renders the auto refill feature useless.