bambulab/BambuStudio #4142: Sound notification when print is paused / blinking PAUSED display

Gruenbaer, 2024-05-22:
**Describe the solution you'd like**
Play a sound when the print is paused.
Display PAUSED blinking on the display.

**Describe alternatives you've considered**
There is no alternative.
Well, there obviously are...surprise me!

**Additional context**
I have a model, which I stop mid print to add weights. I insert a pause in the slicer, start the print.
I watch a movie. 
I do not realize, that the print is paused. 
Play a sound!
I would have realized.
Ain't nobody got time to constantly check the printer. That is sooo 2020s...

Anyhooo, if you consider to do it, also let me choose what is played. midi, mp3, whatever.
Blinking PAUSED display? I`d take that too.

raster, 2024-05-23:
Are you wanting the sound to happen in Bambu Studio or Bambu Handy? I don't believe the printer itself can play any sounds.

Gruenbaer, 2024-05-24:
The prinnter should play the sound. It is capable of playing sounds:
https://wiki.bambulab.com/en/A1-mini/Midi

Gruenbaer, 2024-05-24:
> The printer should play the sound. It is capable of playing sounds: https://wiki.bambulab.com/en/A1-mini/Midi

OK, thking about it I am in the wrong forum, right?