bambulab/BambuStudio #4143: Merge Objects - slicing step

soldino71, 2024-05-23:
### Bambu Studio Version

1.9.1.76

### Where is the application from?

Bambu Lab Official website

### OS version

Sonoma 14.4.1

### Additional system information

MacBook Air M2, 8 GB

### Printer

Bambu A1

### How to reproduce

1 - import 2 STL files (logo and  a rectangle shape)
2 - Color each object using the paint tool.
3 - Ensure both objects are “Lay on Face,” 
4 - merge/assemble them.

### Actual results

Slicing the Plate raises the logo about 2-3 layers above the plate, hiding it, and showing only the card on the first layer (no details are available)

### Expected results

logo should be clearly visible in the first layer

Please consider this process worked fine till I updated the Bambu Studio, one week ago
[BAmbu issue.zip](https://github.com/bambulab/BambuStudio/files/15414000/BAmbu.issue.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/15414074/log.zip)

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/15414077/log.zip)
[BAmbu issue.zip](https://github.com/bambulab/BambuStudio/files/15414078/BAmbu.issue.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-05-28:
[image]
@soldino71  When two parts have different colors and overlap, we give priority to displaying the color later in the object list.

You can drag the blue part behind the white part in the object list

soldino71, 2024-05-28:
Thank you very much: it works

Paolo

closed 2024-05-28 by soldino71