bambulab/BambuStudio #4144: Cannot Assign Material to AMS Bay

rachael7, 2024-05-23:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab X1C

### How to reproduce

1. Install a nozzle and set the new nozzle size/material in the normal way.
2. Load a non-Bambu material in the AMS that does not have a profile for the installed nozzle.
3. Attempt to set the material for the AMS bay through either the printer or Bambu Studio

### Actual results

Materials which do not have a profile for the installed nozzle are not listed in the material dropdown on the touchscreen or in Bambu Studio.

### Expected results

I should be able to tell the printer what material is in each bay, whether or not that material is compatible with the installed nozzle. This came up for me, when after a firmware upgrade, all the AMS material information was reset. One of my materials was not Bambu, but because it wasn't compatible with the present nozzle, I could not set the material type/color.

### Project file & Debug log uploads

[X1C, Adapter, Pneumatic Press.zip](https://github.com/bambulab/BambuStudio/files/15424595/X1C.Adapter.Pneumatic.Press.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/15424621/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

DanBao-Bambu, 2024-05-27:
The materials displayed on Bambu Studio and printer screens are actually extracted information from preset files. The preset information includes the printers (including nozzles) compatible with this material. The system material can be displayed after changing the nozzle information because it exists for each model and nozzle in the system material preset. If you want custom materials to also exist when changing the nozzle model, you need to create the corresponding preset for the corresponding nozzle of that model.
You can click the "+Add Preset" button in Edit Filament to add new presets.

rachael7, 2024-05-28:
Thank you for your reply. I know this information already though. My question regards putting a roll of filament in the AMS that I do not intend to use with the mounted nozzle. I often keep several rolls of filament in the AMS so they will be ready when I need them. Even though I cannot use a particular filament with the installed nozzle, I should still be able to put that roll in the AMS for later use. But I am not given the option because of the way it is set up now. I am requesting that behavior to change, so that I can load any defined filament into the AMS at any time, whether or not it is compatible with the current nozzle.

DanBao-Bambu, 2024-05-29:
@rachael7 Okay, your question has been recorded in the requirement, which needs to be evaluated.

rachael7, 2025-09-07:
> [@rachael7](https://github.com/rachael7) Okay, your question has been recorded in the requirement, which needs to be evaluated.

Hello. Can you please tell me the results of the evaluation? Thank you.