bambulab/BambuStudio #4145: Crash when starting on Arch Linux (AppImage)

tve, 2024-05-23:
### Bambu Studio Version

01.09.01.66

### Where is the application from?

Bambu Lab github releases

### OS version

Arch Linux

### Additional system information

CPU: AMD Ryzen 9 5900HX

### Printer

X1C

### How to reproduce

1. Open BambuStudio
2. wait 10s 'til it crashes

### Actual results

Crash:
```
> bambu-studio --debug=4
[2024-05-23 11:29:34.978578] [0x00007fb49b6b6480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
loading network module at /home/tve/.config/BambuStudio/plugins/libbambu_networking.so
after dlopen, network_module is 0x55804934b250
[2024-05-23 11:29:35.053] [console] [info] TunnelMqtt::reconnect: result: 0

(bambu-studio:721554): Gtk-CRITICAL **: 11:29:35.174: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:721554): Gtk-CRITICAL **: 11:29:35.174: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:721554): GLib-GObject-CRITICAL **: 11:29:35.174: invalid cast from 'wxPizza' to 'GtkCellLayout'

[...]

(bambu-studio:721554): GLib-GObject-CRITICAL **: 11:29:37.786: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:721554): Gtk-CRITICAL **: 11:29:37.786: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
terminate called after throwing an instance of 'nlohmann::detail::type_error'
  what():  [json.exception.type_error.302] type must be boolean, but is null
fish: Job 1, 'bambu-studio --debug=4' terminated by signal SIGABRT (Abort)
```

### Expected results

not crash

### Project file & Debug log uploads

Debug log pasted into actual results above

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2024-05-24:
please send the log in /home/tve/.config/BambuStudio/log here
and also, please delete /home/tve/.config/BambuStudio/system, /home/tve/.config/BambuStudio/BambuStudio.conf , then try again

MilesRdz, 2024-05-26:
[log.zip](https://github.com/bambulab/BambuStudio/files/15448696/log.zip)
I'm running into this as well, I've attached the log and config file.

lanewei120, 2024-05-27:
> /home/tve/.config/BambuStudio/system
@MilesRdz 
thanks
how about delete /home/user.name/.config/BambuStudio/system, then have a try?

MilesRdz, 2024-05-27:
@lanewei120 
That didn't work unfortunately but I  when I start the application I run into a certificate message.
Not sure if that has something to do with it. 
[image]

lanewei120, 2024-05-28:
> @lanewei120 That didn't work unfortunately but I when I start the application I run into a certificate message. Not sure if that has something to do with it. ![Screenshot_20240527_113912](https://private-user-images.githubusercontent.com/3085270/334169025-c81ea466-cdfa-4001-b130-88b83815f976.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY4NjA0MDMsIm5iZiI6MTcxNjg2MDEwMywicGF0aCI6Ii8zMDg1MjcwLzMzNDE2OTAyNS1jODFlYTQ2Ni1jZGZhLTQwMDEtYjEzMC04OGI4MzgxNWY5NzYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUyOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MjhUMDEzNTAzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9N2RmZDZhMDkyYTllOGYyNTViNTQ3ZmRhMGExZGM0NDdjZGY0MDEwMDVkZjM3MjVmNWEzZjcyMjY0NzE0MzY2MSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Vvgi5y30syY-pYaCTt5UyYXB4mPNwRkib1EpSbQtXH0) Edit: Interestingly after I put no instead of yes it seems to start Bambu Studio correctly. I'm not sure if it's a coincidence.

this is a normal popup, which appear at each startup for the first time

do you mean bambustudio bootup successfully each time on youor side?

MilesRdz, 2024-05-30:
> > @lanewei120 That didn't work unfortunately but I when I start the application I run into a certificate message. Not sure if that has something to do with it. ![Screenshot_20240527_113912](https://private-user-images.githubusercontent.com/3085270/334169025-c81ea466-cdfa-4001-b130-88b83815f976.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY4NjA0MDMsIm5iZiI6MTcxNjg2MDEwMywicGF0aCI6Ii8zMDg1MjcwLzMzNDE2OTAyNS1jODFlYTQ2Ni1jZGZhLTQwMDEtYjEzMC04OGI4MzgxNWY5NzYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUyOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MjhUMDEzNTAzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9N2RmZDZhMDkyYTllOGYyNTViNTQ3ZmRhMGExZGM0NDdjZGY0MDEwMDVkZjM3MjVmNWEzZjcyMjY0NzE0MzY2MSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Vvgi5y30syY-pYaCTt5UyYXB4mPNwRkib1EpSbQtXH0) Edit: Interestingly after I put no instead of yes it seems to start Bambu Studio correctly. I'm not sure if it's a coincidence.
> 
> this is a normal popup, which appear at each startup for the first time
> 
> do you mean bambustudio bootup successfully each time on your side now?

Bootup is successful now but I’m not sure why. The only thing I did different was click no on the certificate pop up after deleting the system folder like you suggested.

lanewei120, 2024-05-31:
> please send the log in /home/tve/.config/BambuStudio/log here and also, please delete /home/tve/.config/BambuStudio/system, /home/tve/.config/BambuStudio/BambuStudio.conf , then try again

> please send the log in /home/tve/.config/BambuStudio/log here and also, please delete /home/tve/.config/BambuStudio/system, /home/tve/.config/BambuStudio/BambuStudio.conf , then try again

@tve how about your side, can it bootup successfully after deleting those files

dwolrdcojp, 2024-06-04:
I'm experiencing the same crash. I tried deleting the files like recommended and still can't launch successfully. 

I've attached the log as .txt file

[debug_Tue_Jun_04_11_21_50_10700.txt](https://github.com/user-attachments/files/15566768/debug_Tue_Jun_04_11_21_50_10700.txt)

Bambu Studio Version
[Bambu_Studio_linux_ubuntu-v01.09.01.66.AppImage](https://github.com/bambulab/BambuStudio/releases/download/v01.09.01.66/Bambu_Studio_linux_ubuntu-v01.09.01.66.AppImage)

Where is the application from?
Bambu Lab github releases

OS version
Arch Linux
```sh
~ ❯ uname -r                                                                                 11:29:18 AM
6.9.3-arch1-1
```

Additional system information
CPU: Nvidia 3090 

Printer
X1C

How to reproduce
Open BambuStudio
Press yes or no button, yes crashes no stops proccess

Actual results

```sh
~/Downloads ❯ ./Bambu_Studio_linux_ubuntu-v01.09.01.66.AppImage                              11:27:02 AM
[2024-06-04 11:27:13.737539] [0x0000712fdd285480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
loading network module at /home/max/.config/BambuStudio/plugins/libbambu_networking.so
error, dlerror is /home/max/.config/BambuStudio/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)

(bambu-studio:11474): GLib-GObject-CRITICAL **: 11:27:16.337: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.337: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:11474): GLib-GObject-CRITICAL **: 11:27:16.337: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.337: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:11474): GLib-GObject-CRITICAL **: 11:27:16.338: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.338: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:11474): GLib-GObject-CRITICAL **: 11:27:16.338: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.338: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:11474): GLib-GObject-CRITICAL **: 11:27:16.341: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.341: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:11474): GLib-GObject-CRITICAL **: 11:27:16.341: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.341: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

** (bambu-studio:11474): CRITICAL **: 11:27:16.365: Cannot register URI scheme wxfs more than once

** (bambu-studio:11474): CRITICAL **: 11:27:16.365: Cannot register URI scheme memory more than once

** (bambu-studio:11474): CRITICAL **: 11:27:16.366: Cannot register URI scheme wxfs more than once

[... 30 lines cut ...]
(bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.444: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.444: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.446: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.446: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.450: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.450: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.453: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.453: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.457: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.457: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.461: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.461: gtk_widget_set_size_request: assertion 'width >= -1' failed
zsh: segmentation fault (core dumped)  ./Bambu_Studio_linux_ubuntu-v01.09.01.66.AppImage
```

lanewei120, 2024-06-05:
> I'm experiencing the same crash. I tried deleting the files like recommended and still can't launch successfully.
> 
> I've attached the log as .txt file
> 
> [debug_Tue_Jun_04_11_21_50_10700.txt](https://github.com/user-attachments/files/15566768/debug_Tue_Jun_04_11_21_50_10700.txt)
> 
> Bambu Studio Version [Bambu_Studio_linux_ubuntu-v01.09.01.66.AppImage](https://github.com/bambulab/BambuStudio/releases/download/v01.09.01.66/Bambu_Studio_linux_ubuntu-v01.09.01.66.AppImage)
> 
> Where is the application from? Bambu Lab github releases
> 
> OS version Arch Linux
> 
> ```shell
> ~ ❯ uname -r                                                                                 11:29:18 AM
> 6.9.3-arch1-1
> ```
> 
> Additional system information CPU: Nvidia 3090
> 
> Printer X1C
> 
> How to reproduce Open BambuStudio Press yes or no button, yes crashes no stops proccess
> 
> Actual results
> 
> ```shell
> ~/Downloads ❯ ./Bambu_Studio_linux_ubuntu-v01.09.01.66.AppImage                              11:27:02 AM
> [2024-06-04 11:27:13.737539] [0x0000712fdd285480] [trace]   Initializing StaticPrintConfigs
> add font of HarmonyOS_Sans_SC_Bold returns 1
> add font of HarmonyOS_Sans_SC_Regular returns 1
> loading network module at /home/max/.config/BambuStudio/plugins/libbambu_networking.so
> error, dlerror is /home/max/.config/BambuStudio/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
> after dlopen, network_module is (nil)
> 
> (bambu-studio:11474): GLib-GObject-CRITICAL **: 11:27:16.337: invalid cast from 'wxPizza' to 'GtkCellLayout'
> 
> (bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.337: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> 
> (bambu-studio:11474): GLib-GObject-CRITICAL **: 11:27:16.337: invalid cast from 'wxPizza' to 'GtkCellLayout'
> 
> (bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.337: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> 
> (bambu-studio:11474): GLib-GObject-CRITICAL **: 11:27:16.338: invalid cast from 'wxPizza' to 'GtkCellLayout'
> 
> (bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.338: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> 
> (bambu-studio:11474): GLib-GObject-CRITICAL **: 11:27:16.338: invalid cast from 'wxPizza' to 'GtkCellLayout'
> 
> (bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.338: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> 
> (bambu-studio:11474): GLib-GObject-CRITICAL **: 11:27:16.341: invalid cast from 'wxPizza' to 'GtkCellLayout'
> 
> (bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.341: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> 
> (bambu-studio:11474): GLib-GObject-CRITICAL **: 11:27:16.341: invalid cast from 'wxPizza' to 'GtkCellLayout'
> 
> (bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.341: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
> 
> ** (bambu-studio:11474): CRITICAL **: 11:27:16.365: Cannot register URI scheme wxfs more than once
> 
> ** (bambu-studio:11474): CRITICAL **: 11:27:16.365: Cannot register URI scheme memory more than once
> 
> ** (bambu-studio:11474): CRITICAL **: 11:27:16.366: Cannot register URI scheme wxfs more than once
> 
> ** (bambu-studio:11474): CRITICAL **: 11:27:16.366: Cannot register URI scheme memory more than once
> 
> (bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.377: gtk_window_resize: assertion 'height > 0' failed
> 
> (bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.379: gtk_widget_set_size_request: assertion 'width >= -1' failed
> 
[... 30 lines cut ...]
> (bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.446: gtk_widget_set_size_request: assertion 'width >= -1' failed
> 
> (bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.450: gtk_widget_set_size_request: assertion 'width >= -1' failed
> 
> (bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.450: gtk_widget_set_size_request: assertion 'width >= -1' failed
> 
> (bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.453: gtk_widget_set_size_request: assertion 'width >= -1' failed
> 
> (bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.453: gtk_widget_set_size_request: assertion 'width >= -1' failed
> 
> (bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.457: gtk_widget_set_size_request: assertion 'width >= -1' failed
> 
> (bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.457: gtk_widget_set_size_request: assertion 'width >= -1' failed
> 
> (bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.461: gtk_widget_set_size_request: assertion 'width >= -1' failed
> 
> (bambu-studio:11474): Gtk-CRITICAL **: 11:27:16.461: gtk_widget_set_size_request: assertion 'width >= -1' failed
> zsh: segmentation fault (core dumped)  ./Bambu_Studio_linux_ubuntu-v01.09.01.66.AppImage
> ```

@dwolrdcojp 
your log is different, previous crash is caused by "json.exception" which can be fixed by remove the corrupted json files

was previous version working before?
and could you use gdb to run this appimage and check the callstack then?

dwolrdcojp, 2024-06-05:
@lanewei120 

No, this is first time trying to get this working on Arch for me. 

Here is the trace from gdb

```sh
~/Downloads ❯ gdb ./Bambu_Studio_linux_ubuntu-v01.09.01.66.AppImage                          03:02:32 PM
GNU gdb (GDB) 14.2
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./Bambu_Studio_linux_ubuntu-v01.09.01.66.AppImage...

This GDB supports auto-downloading debuginfo from the following URLs:
  <https://debuginfod.archlinux.org>
Enable debuginfod for this session? (y or [n]) y
Debuginfod has been enabled.
To make this setting permanent, add 'set debuginfod enabled on' to .gdbinit.
(No debugging symbols found in ./Bambu_Studio_linux_ubuntu-v01.09.01.66.AppImage)
(gdb) run
Starting program: /home/max/Downloads/Bambu_Studio_linux_ubuntu-v01.09.01.66.AppImage
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[Detaching after fork from child process 137860]
process 137857 is executing new program: /usr/bin/bash
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[Detaching after fork from child process 137865]
process 137857 is executing new program: /tmp/.mount_Bambu_aiq9hd/bin/bambu-studio
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[2024-06-05 15:02:43.982826] [0x00007fffe38b8480] [trace]   Initializing StaticPrintConfigs
[New Thread 0x7fffe26006c0 (LWP 137869)]
[New Thread 0x7fffe1c006c0 (LWP 137870)]
[Thread 0x7fffe26006c0 (LWP 137869) exited]
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
[New Thread 0x7fffe26006c0 (LWP 137871)]
[New Thread 0x7fffdbe006c0 (LWP 137872)]
[New Thread 0x7fffdb4006c0 (LWP 137873)]
[New Thread 0x7fffda4006c0 (LWP 137874)]
[Detaching after vfork from child process 137944]
[Detaching after vfork from child process 137946]
loading network module at /home/max/.config/BambuStudio/plugins/libbambu_networking.so
error, dlerror is /home/max/.config/BambuStudio/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)

(bambu-studio:137857): GLib-GObject-CRITICAL **: 15:02:52.074: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:137857): Gtk-CRITICAL **: 15:02:52.074: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:137857): GLib-GObject-CRITICAL **: 15:02:52.075: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:137857): Gtk-CRITICAL **: 15:02:52.075: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:137857): GLib-GObject-CRITICAL **: 15:02:52.076: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:137857): Gtk-CRITICAL **: 15:02:52.076: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
[... 68 lines cut ...]
(bambu-studio:137857): Gtk-CRITICAL **: 15:02:52.252: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:137857): Gtk-CRITICAL **: 15:02:52.255: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:137857): Gtk-CRITICAL **: 15:02:52.255: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:137857): Gtk-CRITICAL **: 15:02:52.259: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:137857): Gtk-CRITICAL **: 15:02:52.259: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:137857): Gtk-CRITICAL **: 15:02:52.263: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:137857): Gtk-CRITICAL **: 15:02:52.263: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:137857): Gtk-CRITICAL **: 15:02:52.267: gtk_widget_set_size_request: assertion 'width >= -1' failed

(bambu-studio:137857): Gtk-CRITICAL **: 15:02:52.267: gtk_widget_set_size_request: assertion 'width >= -1' failed

Thread 1 "bambustu_main" received signal SIGSEGV, Segmentation fault.
0x0000555557142162 in wxMediaCtrl2::wxMediaCtrl2(wxWindow*) ()
(gdb) kill
Kill the program being debugged? (y or n) y
[Inferior 1 (process 137857) killed]
(gdb)
```

lanewei120, 2024-06-06:
could you use command 'bt' before 'kill' under gdb?
this will show the full call stack

you can use the flathub version here https://flathub.org/apps/com.bambulab.BambuStudio

dwolrdcojp, 2024-06-06:
```sh
~/Downloads ❯ gdb ./Bambu_Studio_linux_ubuntu-v01.09.01.66.AppImage                          11:18:55 AM
GNU gdb (GDB) 14.2
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./Bambu_Studio_linux_ubuntu-v01.09.01.66.AppImage...

This GDB supports auto-downloading debuginfo from the following URLs:
  <https://debuginfod.archlinux.org>
Enable debuginfod for this session? (y or [n]) y
Debuginfod has been enabled.
To make this setting permanent, add 'set debuginfod enabled on' to .gdbinit.
Downloading separate debug info for /home/max/Downloads/Bambu_Studio_linux_ubuntu-v01.09.01.66.AppImage
(No debugging symbols found in ./Bambu_Studio_linux_ubuntu-v01.09.01.66.AppImage)
(gdb) run
Starting program: /home/max/Downloads/Bambu_Studio_linux_ubuntu-v01.09.01.66.AppImage
Downloading separate debug info for system-supplied DSO at 0x7ffff7fc7000
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
Downloading separate debug info for /usr/lib/libfuse.so.2
[Detaching after fork from child process 5960]
process 5957 is executing new program: /usr/bin/bash
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[Detaching after fork from child process 5965]
process 5957 is executing new program: /tmp/.mount_Bambu_mlG8Fy/bin/bambu-studio
Downloading separate debug info for /tmp/.mount_Bambu_mlG8Fy/bin/bambu-studio
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
Downloading separate debug info for /usr/lib/libtasn1.so.6
[2024-06-06 11:19:09.789048] [0x00007fffe38b8480] [trace]   Initializing StaticPrintConfigs
[New Thread 0x7fffe26006c0 (LWP 5973)]
[New Thread 0x7fffdbe006c0 (LWP 5974)]
[Thread 0x7fffe26006c0 (LWP 5973) exited]
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
Downloading separate debug info for /usr/lib/libvorbisfile.so.3
Downloading separate debug info for /usr/lib/libvorbis.so.0
Downloading separate debug info for /usr/lib/libogg.so.0
[New Thread 0x7fffe26006c0 (LWP 5975)]
[New Thread 0x7fffe0c006c0 (LWP 5976)]
[New Thread 0x7fffdb4006c0 (LWP 5977)]
Downloading separate debug info for /usr/lib/libvorbisenc.so.2
[New Thread 0x7fffda4006c0 (LWP 5978)]
[Detaching after vfork from child process 6019]
[Detaching after vfork from child process 6021]
loading network module at /home/max/.config/BambuStudio/plugins/libbambu_networking.so
error, dlerror is /home/max/.config/BambuStudio/plugins/libbambu_networking.so: cannot open shared object file: No such file or directory
after dlopen, network_module is (nil)

(bambu-studio:5957): GLib-GObject-CRITICAL **: 11:19:12.963: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:5957): Gtk-CRITICAL **: 11:19:12.963: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(bambu-studio:5957): GLib-GObject-CRITICAL **: 11:19:12.964: invalid cast from 'wxPizza' to 'GtkCellLayout'

(bambu-studio:5957): Gtk-CRITICAL **: 11:19:12.964: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

[... 93 lines cut ...]

Thread 1 "bambustu_main" received signal SIGSEGV, Segmentation fault.
0x0000555557142162 in wxMediaCtrl2::wxMediaCtrl2(wxWindow*) ()
(gdb) bt
#0  0x0000555557142162 in wxMediaCtrl2::wxMediaCtrl2(wxWindow*) ()
#1  0x00005555570e23a5 in Slic3r::GUI::StatusBasePanel::create_monitoring_page() ()
#2  0x00005555570e6b14 in Slic3r::GUI::StatusBasePanel::StatusBasePanel(wxWindow*, int, wxPoint const&, wxSize const&, long, wxString const&) ()
#3  0x00005555570eff0a in Slic3r::GUI::StatusPanel::StatusPanel(wxWindow*, int, wxPoint const&, wxSize const&, long, wxString const&) ()
#4  0x0000555556b8ac2d in Slic3r::GUI::MonitorPanel::init_tabpanel() ()
#5  0x0000555556b8b3d6 in Slic3r::GUI::MonitorPanel::MonitorPanel(wxWindow*, int, wxPoint const&, wxSize const&, long) ()
#6  0x0000555556984dc5 in Slic3r::GUI::MainFrame::init_tabpanel() ()
#7  0x000055555698612e in Slic3r::GUI::MainFrame::MainFrame() ()
#8  0x000055555693def1 in Slic3r::GUI::GUI_App::on_init_inner() ()
#9  0x000055555693ef5a in Slic3r::GUI::GUI_App::OnInit() ()
#10 0x0000555559562607 in wxEntry(int&, wchar_t**) ()
#11 0x000055555690010b in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#12 0x0000555555f67512 in Slic3r::CLI::run(int, char**) ()
#13 0x0000555555e7f129 in main ()
(gdb) kill
Kill the program being debugged? (y or n) y
[Inferior 1 (process 5957) killed]
(gdb) exit
```

Thanks I will look into the flathub version

lanewei120, 2024-06-07:
thanks, it is very likely caused by gstreamer related libraries