bambulab/BambuStudio #4146: Crashing when startup on Windows 11

fnoller, 2024-05-24:
### Bambu Studio Version

1.09.01.67

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 11

### Additional system information

CPU: AMD Ryzen 9 7950X3D 16-Core Processor
MB: ROG CROSSHAIR X670E Hero
DRAM: 128GB
GPU: ROG-Strix-RTX-4090

### Printer

Bambu Lab P1P

### How to reproduce

Start the program, see the splash windows and the program starts up for 2 seconds and then shutdown.

I have tried with the versions:

1.09.01.66
1.09.01.67
1.09.01.6 Beta
1.09.01.6 Beta2
1.09.01.6 Beta3
1.09.01.6 Beta4

All with the same result. 

### Actual results

The program starts up for 2 seconds and then shutdown.

### Expected results

The program was running.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/15428768/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Haidiye00, 2024-05-29:
Sorry,please wait for the next version

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

closed 2025-04-16 by Haidiye00