bambulab/BambuStudio #4148: 4060显卡下，拖拽整个软件界面，延迟巨大

wangzejie1108, 2024-05-24:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

win10

### Additional system information

CPU2700X，显卡4060

### Printer

p1scombo

### How to reproduce

拖动整个界面就好了，很大延迟，不跟着鼠标走，或者说，鼠标响应很长时间

### Actual results

链接：https://pan.baidu.com/s/1E40BBtsjjM7u6OzQAHp2rQ?pwd=GanV 
提取码：GanV

### Expected results

11111

### Project file & Debug log uploads

无

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2024-05-27:
这个问题我们一直在研究，问题很奇怪，目前还没有找到原因。

phranck, 2024-05-27:
It would be great if all issues and pull requests were only written in English. Then everyone would have the chance to participate without having to have them translated first.

Thanks.