bambulab/BambuStudio #4150: P1S always listed in Bambu Studio

DanlB111, 2024-05-25:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Built from source by yourself or others

### OS version

Windows 11

### Additional system information

12th Gen Intel(R) Core(TM) i9-12900   2.40 GHz
64.0 GB (63.7 GB usable)
NVIDIA GeForce RTX 3080

### Printer

Bambu Labs p1S

### How to reproduce

Every time I open Bambu Studio.

### Actual results

P1S is selected in the printer choice even though I have a X1 Carbon and P1S is not selected in the printer choices window.

### Expected results

I should only have 1X C as the printer in the box.

### Project file & Debug log uploads

[Scoop.zip](https://github.com/bambulab/BambuStudio/files/15441054/Scoop.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

DanlB111, 2024-05-25:
> If you are loading the scoop.3mf file "Every time I open Bambu Studio.",
> this is not a bug. It is intentional so that Studio has settings that are
> known to work. Either change the printer after loading the .3mf or just use
> the "Import geometry only" option.

thx for the reply.

*Dan Bernskoetter Photography*
***@***.*** ***@***.***>*
https//www.MoOutdoorPhotography.com <http://www.MoOutdoorPhotography.com>

lanewei120, 2024-05-27:
> If you are loading the scoop.3mf file "Every time I open Bambu Studio.", this is not a bug. It is intentional, so that Studio has settings that are known to work. Either change the printer after loading the .3mf or just use the "Import geometry only" option.

yes, correctly~
not a bug

closed 2024-05-27 by lanewei120