bambulab/BambuStudio #4152: Color information lost after obj import dialog box completes

timperlabs, 2024-05-25:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 13.5

### Additional system information

NA

### Printer

X1C

### How to reproduce

1.  Create a model in Fusion360 with multiple bodies with different colors
2. Export as .OBJ to Bambu Studio
3. Choose a mapping to multiple filaments in the color mapping dialog box and click OK.

### Actual results

The model is display in a single color.

### Expected results

I expect to see the model in multiple colors as specified in the dialog box.

By going to "Objects" tab and then "Split to Parts" on the main .obj, you can manually fix the mapping.  I have included screenshots of all the steps starting from reproducing the error, observing the error, and then manually fixing it.

[image]
[image]
[image]
[image]
[image]

### Project file & Debug log uploads

[ColorBug.zip](https://github.com/bambulab/BambuStudio/files/15444128/ColorBug.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

ttt888zzz, 2024-05-27:
Thank you for your feedback
Could you upload the original OBJ file

Haidiye00, 2024-05-31:
Could you upload the original OBJ file,thanks

Haidiye00, 2025-03-26:
Hello, we have released software V2.0. Please download and use it
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.00.95

closed 2025-04-16 by Haidiye00