bambulab/BambuStudio #4155: Port OrcaSlicer precise walls feature.

akravetz, 2024-05-25:
**Is your feature request related to a problem? Please describe.**
Precise wall is a feature in the open source project orcaslicer. It yields a notable visible difference in print quality without affecting the structure. 

See details here: https://github.com/SoftFever/OrcaSlicer/discussions/831

**Describe the solution you'd like**
I am happy to put up a PR for this port if bambu takes outside contributions.

**Describe alternatives you've considered**
Right now I've stopped using the bambu slicer for high quality prints because orca is so much better with the precise wall feature

**Additional context**

relax81, 2024-07-26:
+1 for this. 

This feature also helps a lot with layer shifts when the print has a complete bottom infill and then transitions to something like a 3mm outer wall.  Such situations are sometimes hard to "tame" with Bambu Studio, but "precise wall" of Orca Slicer helps a lot to reduce that issue to a point where its barely visible.  

Would be very nice to have this in Bambu Studio so we could use it in our MakerWorld Print profiles.

DrFischie, 2024-12-16:
And we're still waiting.