bambulab/BambuStudio #4157: Arachne wall generator can start extrusion in the middle of an overhang

ReneZeidler, 2024-05-26:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11, 23H2

### Printer

Bambu Lab X1-Carbon

### How to reproduce

1. Open the included project file
2. Slice with 0.20 mm layer height, 3 wall loops and the Arachne wall generator (should be pre-configured)
3. Observe layer 157/height 31.40 mm - the extrusion path starts in the middle of an overhang/a bridge.

### Actual results

[image]
[image]
The extrusion path starts in the middle of a bridge, which obviously won't work.

### Expected results

The Arachne wall generator shouldn't create even-odd wall transitions over 100% overhangs, because those can result in impossible extrusion paths.

### Project file & Debug log uploads

Project File: [Floating Wall.zip](https://github.com/bambulab/BambuStudio/files/15445703/Floating.Wall.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

PerfectDark64, 2025-04-20:
Issue is still present in latest Bambu Studio 2.0.2.57. Issue also seems to crop up in Classic if layer height is set to 0.08mm

[image]

XunZhangBambu, 2025-04-30:
@PerfectDark64 @ReneZeidler  Our developers have started analyzing this issue. This could be a large change so we need some time