bambulab/BambuStudio #4158: occt cant build

echo-lxq, 2024-05-26:
### Bambu Studio Version

last version

### Where is the application from?

Bambu Lab github releases

### OS version

window 10

### Additional system information

[image]

### Printer

x1 series

### How to reproduce

building the deps

### Actual results

已启动生成…
1>------ 已启动生成: 项目: dep_FREETYPE, 配置: Debug x64 ------
2>------ 已启动生成: 项目: dep_OCCT, 配置: Debug x64 ------
2>Performing patch step for 'dep_OCCT'
2>Checking patch deps/build/dep_OCCT-prefix/src/dep_OCCT/CMakeLists.txt...
2>CUSTOMBUILD : error : deps/build/dep_OCCT-prefix/src/dep_OCCT/CMakeLists.txt: No such file or directory
2>Checking patch deps/build/dep_OCCT-prefix/src/dep_OCCT/src/Font/Font_FTFont.cxx...
2>CUSTOMBUILD : error : deps/build/dep_OCCT-prefix/src/dep_OCCT/src/Font/Font_FTFont.cxx: No such file or directory
2>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(241,5): error MSB8066: “D:\work\Projects\BambuStudio\deps\build\CMakeFiles\433f5f2e4fd2cfbe4738a5a96451353e\dep_OCCT-patch.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\433f5f2e4fd2cfbe4738a5a96451353e\dep_OCCT-configure.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\433f5f2e4fd2cfbe4738a5a96451353e\dep_OCCT-build.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\433f5f2e4fd2cfbe4738a5a96451353e\dep_OCCT-install.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\68c521bc805cc530064eef33e47b39d2\dep_OCCT-complete.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\258fc9b4ab6c35b7f3020ecde013b882\dep_OCCT.rule”的自定义生成已退出，代码为 1。
2>已完成生成项目“dep_OCCT.vcxproj”的操作 - 失败。
========== 生成: 成功 1 个，失败 1 个，最新 1 个，跳过 0 个 ==========

### Expected results

 cant build

### Project file & Debug log uploads

已启动生成…
1>------ 已启动生成: 项目: dep_FREETYPE, 配置: Debug x64 ------
2>------ 已启动生成: 项目: dep_OCCT, 配置: Debug x64 ------
2>Performing patch step for 'dep_OCCT'
2>Checking patch deps/build/dep_OCCT-prefix/src/dep_OCCT/CMakeLists.txt...
2>CUSTOMBUILD : error : deps/build/dep_OCCT-prefix/src/dep_OCCT/CMakeLists.txt: No such file or directory
2>Checking patch deps/build/dep_OCCT-prefix/src/dep_OCCT/src/Font/Font_FTFont.cxx...
2>CUSTOMBUILD : error : deps/build/dep_OCCT-prefix/src/dep_OCCT/src/Font/Font_FTFont.cxx: No such file or directory
2>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(241,5): error MSB8066: “D:\work\Projects\BambuStudio\deps\build\CMakeFiles\433f5f2e4fd2cfbe4738a5a96451353e\dep_OCCT-patch.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\433f5f2e4fd2cfbe4738a5a96451353e\dep_OCCT-configure.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\433f5f2e4fd2cfbe4738a5a96451353e\dep_OCCT-build.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\433f5f2e4fd2cfbe4738a5a96451353e\dep_OCCT-install.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\68c521bc805cc530064eef33e47b39d2\dep_OCCT-complete.rule;D:\work\Projects\BambuStudio\deps\build\CMakeFiles\258fc9b4ab6c35b7f3020ecde013b882\dep_OCCT.rule”的自定义生成已退出，代码为 1。
2>已完成生成项目“dep_OCCT.vcxproj”的操作 - 失败。
========== 生成: 成功 1 个，失败 1 个，最新 1 个，跳过 0 个 ==========

### Checklist of files to include

- [X] Log file
- [X] Project file

MackBambu, 2024-05-30:
In my last few compilation tests, all sorts of surprises occurred when memory ran low.
It's also very important to maintain good network conditions.

JuergenKammer, 2024-06-20:
Got the same issue, on linux debian. Solved it by changing the PATCH_COMMAND in file OCCT.cmake, located in directory deps/OCCT from

`PATCH_COMMAND git apply --directory deps/build/dep_OCCT-prefix/src/dep_OCCT --verbose --ignore-space-change --whitespace=fix ${CMAKE_CURRENT_LIST_DIR}/0001-OCCT-fix.patch`

to

`PATCH_COMMAND ${GIT_EXECUTABLE} apply --verbose --ignore-space-change --whitespace=fix ${CMAKE_CURRENT_LIST_DIR}/0001-OCCT-fix.patch`

After this change the complete build for the deps, bambu-studio and appimage ran through without problem.

MackBambu, 2024-06-21:
@JuergenKammer Thanks, I'll merge the changes.

closed 2024-07-17 by MackBambu

phantasyconcepts, 2025-01-30:
It looks like this issue is back in the 1.10.01.50 source code.  I had to make the same change as @JuergenKammer suggested to get past that on CentOS 9 Stream.  Incidentally, the 'buildLinux' scripts need to be modified to respond to CentOS as Fedora as well.  I can create a new issue for this, but it looks like someone reintroduced the issue.