bambulab/BambuStudio #4159: 13" monitor, the "send print to the printer" windows has the "send"button covered by the windows bar

terxo, 2024-05-27:
### Bambu Studio Version

the last one

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

bambu printer

### How to reproduce

anytime when i send a print with my laptop

### Actual results

13" monitor on my laptop.
the "send print to the printer" window have the "send" button covered by the windows bar.

So i'm obliged to use the auto hide windows bar.... but unother problem comes.. when bambustudio is fullscreen scaled  the windows bar is unavailable when i go with my mouse

so:
to send brint i need the windows bar autohide
but anytime i'm using bambustudio i need to resize to non-fullscreen mode to be able to use my windows bar..

it's frustrating to do that so nany times a day

### Expected results

13" monitor on my laptop.
the "send print to the printer" window have the "send" button covered by the windows bar.

So i'm obliged to use the auto hide windows bar.... but unother problem comes.. when bambustudio is fullscreen scaled  the windows bar is unavailable when i go with my mouse

so:
to send brint i need the windows bar autohide
but anytime i'm using bambustudio i need to resize to non-fullscreen mode to be able to use my windows bar..

it's frustrating to do that so nany times a day

### Project file & Debug log uploads

issue indipendent from the project and printer

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ttt888zzz, 2024-05-28:
Thank you for your feedback.
May I ask if you can provide a screenshot
Thank you.

terxo, 2024-05-28:
> Links:

You're wellcome!

Screenshots attached!

if windows bar is visible, no way to print because the send print button 
is behind the bar

so i use the auto hidden windows bar... but... if bambustudio is full 
screen the mouse trigger to wakeup the windows bar it doesn't work!

Thank you.
Simone

ttt888zzz, 2024-05-29:
1. Modification: Settings - Display Resolution, Scale
2. Pictures related to my computer settings:
[image]

[image]

terxo, 2024-09-17:
Hello.

I can not use bambustudio anymore, i can not read any command/value. see 
the picture.
what can it be, Wehat can I do?

Thank you

ttt888zzz, 2024-09-18:
HI，[terxo](https://github.com/terxo)：The file you uploaded cannot be seen.
Please reproduce your problem scenario by uploading a screenshot or recording a screen.
[image]

terxo, 2024-09-18:
impossible to see any command writing in bambustudio

[image]

[image]

[image]

bambu123, 2024-09-19:
Try switching to dark mode in preference.

terxo, 2024-09-20:
"Try switching to dark mode in preference"

impossible to see anything in preferences

bambu123, 2024-09-20:
Please provide log files.
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

> From: "Simone ***@***.***>
> Date:  Fri, Sep 20, 2024, 18:19
> Subject:  Re: [bambulab/BambuStudio] 13" monitor, the "send print to the printer" windows has the "send"button covered by the windows bar (Issue #4159)
> To: ***@***.***>
> Cc: ***@***.***>, ***@***.***>
> 
"Try switching to dark mode in preference"
> 
> 
impossible to see anything in preferences
> 
> 
>