bambulab/BambuStudio #4163: bambu studio or PC crashes when slicing

Phil-2310, 2024-05-27:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro

### Additional system information

CPU: 13th Gen Intel(R) Core(TM) i7-13700K   3.40 GHz
Ram: 32GB
Nvidia GeForce RTX 2070 SUPER
64-bit OS

### Printer

Bambu A1 mini

### How to reproduce

Happens when trying to slice 0.06mm layer height (sometimes 0.08mm) and noticeably when trying to use tree support. Tried with several models and didn't happen when using older software version. Slices OK when using larger layer heights.

### Actual results

Normally Bambu Studio crashes, sometimes whole PC

### Expected results

Slice the file

### Project file & Debug log uploads

no file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2024-05-28:
@Phil-2310  Could you share your crash log, or or it would be better if there is a dump file?

zhimin-zeng-bambulab, 2024-08-09:
@Phil-2310 Or could you share your 3mf file?

deadpin, 2024-11-10:
Having the same issue here. I've found two .stl/3mf files doing it and pin pointed it down to being 0.06mm layer height using a 0.2nozzle. Adjusting layer height to 0.08 allows slicing to process beyond the generating walls phase. 
[bambu-studio.exe.12548.dmp](https://github.com/user-attachments/files/17693615/bambu-studio.exe.12548.dmp)
[crash_Sun_Nov_10_18_57_01_0.log](https://github.com/user-attachments/files/17693616/crash_Sun_Nov_10_18_57_01_0.log)