bambulab/BambuStudio #4165: Publish the network plugin for more architecture and OS combinations

MisterTechnik, 2024-05-28:
**Is your feature request related to a problem? Please describe.**
I successfully built a fork of BambuStudio, more specific OrcaSlicer, for Ubuntu on ARM, but I cannot use it reasonably for my A1 Mini without the libbambu_networking.so for ARM64 architecture.

**Describe the solution you'd like**
This feature request [#2381](https://github.com/bambulab/BambuStudio/issues/2381) would be the ideal solution.
But until you get this figured out, can you please publish this and the other precompiled libraries for ARM on Linux and probably Windows, too?

**Describe alternatives you've considered**
Emulating a single x86 library for ARM64 to work with the Slicer is above my skill level, if even possible. The same applies to reverse engineering and recompiling for ARM which would also probably be very much against your intentions with these files.

**Additional context**
_emptiness_

leggazoid, 2024-07-13:
libbambu_networking.so for aarch64 would allow raspberry pi to run bambu studio.

liweitianux, 2025-11-16:
Second to this. I'd really like this network plugin comes to FreeBSD, DragonFly and other BSDs.