bambulab/BambuStudio #4166: [CVE-2022-25235] Dependency libexpat version does not protect against malformed UTF-8

Nyhtfury, 2024-05-28:
### Bambu Studio Version

1.9.1

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10 & 11

### Printer

Bambu Lab X1E (proposed)

### How to reproduce

1. Look at the GitHub repository and search for xmltok_impl.c
2. There are two file versions (with file extension .inc), one under /src/expat, and another under /deps/EXPAT/expat.
3. The deps version is much newer (2022) compared to the src version (1999). The newer version matches what is available in the [libexpat repository](https://github.com/libexpat/libexpat/blob/master/expat/lib/xmltok_impl.c).

See the CVE here: https://www.mend.io/vulnerability-database/CVE-2022-25235

### Actual results

My IT department will not allow this program with a critical remote execution vulnerability due to the old version of Expat (i.e. libexpat) that Bambu Studio is being built with. As a result, a printer is not being purchased.

### Expected results

To resolve the RCE CVE, Expat (i.e. libexpat) must be updated to at least 2.4.5. I would also expect the code base to standardize on a version.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-05-29:
Please confirm whether only libexpat needs to be updated due to the CVE-xxx

Nyhtfury, 2024-05-29:
> Please confirm whether only libexpat needs to be updated due to the CVE-xxx

@lanewei120 per the CVE: "xmltok_impl.c in Expat (aka libexpat) before 2.4.5 lacks certain validation of encoding, such as checks for whether a UTF-8 character is valid in a certain context."

Thus, my understanding is that only libexpat needs to be updated to address CVE-2022-25235.

MackBambu, 2024-06-04:
@Nyhtfury I will try to update Expat

MackBambu, 2024-11-24:
fixed

closed 2024-11-24 by MackBambu