bambulab/BambuStudio #4168: Manual Flow Dynamics Calibration - Message informing about historic data storaf

JDPS, 2024-05-28:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU :Intel Core i5-9600K
MB: MSI Z390-A PRO (MS-7B98)
RAM: 48Gb  2xF4-3000C16-16GISB + 2xF4-3000C8-8GISB
GPU: GIGABYTE GTX 1660 SUPER OC 6G (GV-N166SOC-6GD)
HD:  Samsung SSD 970 EVO Plus 1TB

### Printer

Bambu Lab X1 Carbon Combo

### How to reproduce

1. Select (left mouse click)  the Calibration Tab on the top bar.
2. Select (left mouse click)  flow dynamics on the left bar;
3. Select (left mouse click) the manual calibration button;
4. Select (left mouse click)  the calibration button;

### Actual results

Receive a pop-up message with the following content:

"This machine type can only hold 16 historical results per nozzle. You can delete the existing
historical results and then start calibration. Or you can continue the calibration, but you cannot
create new calibration historical results. 
Do you still want to continue the calibration?"
[image]

If I proceed, the process will work as expected, but the message will seem pointless and incorrect.

### Expected results

Start the calibration procedure without any pop-up message.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/15474241/log.zip)
I am not doing any project.

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-05-29:
@JDPS 
This message prompts you that you cannot store new calibration values ​​in the management table. 
If you choose to calibrate, you can continue, but your calibration values ​​will not be stored in the management list.

JDPS, 2024-05-29:
@zhimin-zeng-bambulab, many thanks for the swift answer.
I found this issue after a user complaint in the BL forum.
I could indeed replicate and got the same message, but something doesn't seem right:

- I  have at least double the k-values memorised in the management table for the 0.4mm nozzle.
- If I choose the auto flow dynamics calibration, no alert message will be shown, and the values will be added to the existing ones.
- Whithin the management table, I can press the new button, allowing me to add a new filament and respective k-avlue, without any alert. 

So, the alert message only shows when using a manual calibration procedure.
 Additionally, by ignoring the alert and continuing with the calibration procedure, I successfully saved the new filament calibration values. I did it three times successfully. 

To ensure we are discussing the same thing, I assume the "calibration management table" relates to the table shown when one selects the option "manage results". Correct?

Thank you.

zhimin-zeng-bambulab, 2024-05-29:
@JDPS  Thanks for your patient explanation.   X1C should not have this restriction. This is a BUG. We will fix it. Thanks!

JDPS, 2024-05-29:
Thanks for the clarification, and I'm glad it can help.

zhimin-zeng-bambulab, 2024-05-29:
@JDPS  We will notify you here when the fixed version comes out

zhimin-zeng-bambulab, 2024-06-12:
@JDPS  It has been fixed in v01.09.02.57, please have a try.

closed 2024-06-12 by zhimin-zeng-bambulab

oekobauer, 2024-10-24:
Why isn't the limitation generally lifted for all printers? It makes no sense to do it only for the X1 series and not for the A or P series. Users of the P1S / P1P series in particular would also benefit from this, as even private users usually have more than 16 filaments. So why the unnecessary restriction?

oekobauer, 2024-10-25:
> Why isn't the limitation generally lifted for all printers? It makes no sense to do it just for the X1 series and not for the A or P series. Users of the P1S / P1P series in particular would also benefit from this, as even private users usually have more than 16 filaments. So why the unnecessary restriction? You could enter the PA / K values ​​directly into the filaments, like with Orca, and transfer them accordingly in the G code, or only transfer the values ​​to the system when the corresponding filament is loaded. This could easily be done using a local database on the PC, i.e. the K or PA values ​​from the calibration list are only sent to the printer when the corresponding filaments are loaded into the AMS.

zhimin-zeng-bambulab, 2024-10-28:
1. Due to the small memory of the P/A series machines, the data that can be stored is limited.
2. The k value is related to both the material and the machine. If it is placed in the material preset, it is not universal.