bambulab/BambuStudio #4169: OBJ file with color information creates abnormal internal geometry when sliced

PanamonCreel, 2024-05-29:
### Bambu Studio Version

1.9.1.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab P1S

### How to reproduce

1. Add OBJ file part with color information
2.  confirm Color Assignments
3. In case of the supplied OBJ file resize 1000%
4. Slice part 
5. Scroll through Layers in the Preview Window to see Irregular internal geometries and overlapping colors
6. Return to prepare screen, Process-Objects select part and split into objects
7. Apply different colors to the objects since color information from OBJ file is lost when splitting
8. Slice and scroll through layers,  internal color geometry is correctly translated

### Actual results

Slicing the OBJ file with color info will create irregular internal geometries and color overlaps 
[image]

Note issue has also been reported at the Bambu Labs Forum by another user : https://forum.bambulab.com/t/colors-crossing-over-when-slicing/78075

### Expected results

What it should look like when sliced:
[image]

### Project file & Debug log uploads

[debug_Tue_May_28_13_21_26_3568.log.zip](https://github.com/bambulab/BambuStudio/files/15477402/debug_Tue_May_28_13_21_26_3568.log.zip)
[multiComponentTestFusion_ v3_Report.zip](https://github.com/bambulab/BambuStudio/files/15477407/multiComponentTestFusion_.v3_Report.zip)
[multiComponentTestFusion_ v3.zip](https://github.com/bambulab/BambuStudio/files/15477408/multiComponentTestFusion_.v3.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-05-29:
@PanamonCreel 
The internal area division of coloring is based on a fixed algorithm.
Is there anything wrong with this division method? It will not affect your surface color.

PanamonCreel, 2024-06-17:
@zhimin-zeng-bambulab

Can you please explain to  me the reason for this "intrusive"  fixed Algorithm, what's the thought behind it?

zhimin-zeng-bambulab, 2024-06-18:
@PanamonCreel  It can increase the adhesion between multi-color parts. If multi-colored parts are too flush with each other, they won’t bond well enough

Cryghast, 2026-01-15:
我觉得这是个bug
I think it is a bug