bambulab/BambuStudio #4173: There are no compatible filaments, and sync is not performed.

blhof, 2024-05-29:
### Bambu Studio Version

Newest

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1 Carbon

### How to reproduce

1. Open in makerworld from website
2. Sync AMS
3. There are no compatible filaments, and sync is not performed. is the error I get

### Actual results

There are no compatible filaments, and sync is not performed.

### Expected results

To have it sync

### Project file & Debug log uploads

https://makerworld.com/en/models/248062?from=search#profileId-264283

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tjc2005, 2024-05-30:
Yep, getting this constantly.

daolis, 2024-06-05:
Getting this too. Any updates on this issue?

tjc2005, 2024-06-18:
It happens all the time I've noticed since the new bambu update!!!

DanBao-Bambu, 2024-06-21:
Please provide me with the log at the time of your operation, for reference: https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

blhof, 2024-07-07:
No help on this, guess time to switch to a different printer company.

tjc2005, 2024-07-07:
This is happening all the time still. Unless I close bambustudio constantly and reopen. Useless. This is ever since the latest studio update. Fix it please. 

I like to have multiple studio instances open this is ridiculous that nobody is talking about this more.

scubar, 2024-08-20:
I have an A1 with AMS and an A1 Mini without AMS. I was getting this error constantly trying to use TPU in my A1 Mini while my A1 AMS is only loaded with PLA and PETG.

I disabled multi-device management and was able to perform a sync from the prepare tab again.

TinyMito, 2024-08-26:
> Found a solution that works for me. I just select the printer from the drop down menu twice. Don't go back to the "Prepare" tab, just click the printer in the drop-down menu on the "Device" tab, then select the same printer again.
> 
> Then go to the "Prepare" tab and sync and it works for me.

I came here to confirm this workaround. A quite annoying bug.

iplayfast, 2024-10-17:
linux version 1.9.7.52 still has it, and the beta preview is even stupider. The sync button is gone.

econwaygit, 2024-11-28:
The issue persists (1.9.7.52 - not interested in beta testing newer versions).

The workaround to 'select the device again in prepare tab' worked for me which should provide more than enough info to solve this problem.  Also, I usually have more than one instance open so perhaps that contributes.

piotr-kubica, 2025-02-04:
Don't buy the X1C. I'm getting this constantly on random occasions. What is this?? Totally disappointed!