bambulab/BambuStudio #4175: Supports generating off of bed / palte

EhRabz, 2024-05-30:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab github releases

### OS version

win11

### Additional system information

12900k , 3080ti , z690 , 32gb ram 

### Printer

P1s

### How to reproduce

Slice an object that takes up the whole bed and the supports will generate outside the limits

[image]

### Actual results

[image]

### Expected results

slices WITHIN plate limits.

### Project file & Debug log uploads

[Helldivers BO-1 Helmet.zip](https://github.com/bambulab/BambuStudio/files/15491966/Helldivers.BO-1.Helmet.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

EhRabz, 2024-05-30:
[image]
when using 1.8 beta , you get the proper results while opening the SAME project, no changes. This is 100% a BBLS issue.

EhRabz, 2024-05-30:
[image]
1.8.45.1 is fine also

EhRabz, 2024-05-30:
1.9.1 Public Beta 4 is the LAST one with expected results.

EhRabz, 2024-05-30:
it starts to break at 1.9 beta

RevanitaPL, 2024-06-11:
I created account, only to tell, that this bug is making me using only Bambu Studio 1.8.3, because they still don't fix it.

EhRabz, 2024-06-12:
> I created account, only to tell, that this bug is making me using only Bambu Studio 1.8.3, because they still don't fix it.

Same... And there was even an update today , very sad to see such an important CORE feature is broken.

tangketan, 2024-06-17:
The default tree support tyle has changed to "tree organic", which doesn't have the feature "clapping support at plate boundary" yet. You can manually set the support style to "tree hybrid" in this case.

EhRabz, 2024-10-04:
Still unadressed