bambulab/BambuStudio #4185: Crashing when Slicing Babu Studio Windows 11 5/30/2024

rhuisking, 2024-05-31:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Home 22631.3593

### Additional system information

13th Gen Intel Core i7-13700KF
32.0 GB DDR5
GeForce RTX 3070Ti

### Printer

Bambu Lab A1 Mini

### How to reproduce

Slicing any file on Bambu Studio causes a crash half way through slicing.

### Actual results

Bambu Studio Crash

### Expected results

Should slice the files.

### Project file & Debug log uploads

[Glock+19+v6.zip](https://github.com/user-attachments/files/15507394/Glock%2B19%2Bv6.zip)
[crash_Thu_May_30_14_18_12_0_done.log](https://github.com/user-attachments/files/15507401/crash_Thu_May_30_14_18_12_0_done.log)
[crash_Thu_May_30_17_07_35_0_done.log](https://github.com/user-attachments/files/15507402/crash_Thu_May_30_17_07_35_0_done.log)
[crash_Thu_May_30_17_21_33_0_done.log](https://github.com/user-attachments/files/15507403/crash_Thu_May_30_17_21_33_0_done.log)
[crash_Thu_May_30_18_29_13_0.log](https://github.com/user-attachments/files/15507405/crash_Thu_May_30_18_29_13_0.log)

Other stl file unable to upload due to high file size.

Thanks,
Ryan

### Checklist of files to include

- [X] Log file
- [X] Project file

QingZhangBambu, 2024-05-31:
https://github.com/bambulab/BambuStudio/issues/3318  These methods may help. The CPU may not be stable enough to support multithreading.

QingZhangBambu, 2024-06-03:
and could you upload dump files if the these not work

MackBambu, 2024-06-04:
Can you tell me the model of your motherboard and whether it has functions such as automatic overclocking enabled?

rhuisking, 2024-06-04:
That fixed it. It was the AI overlocking.
Thank you!

closed 2024-06-05 by QingZhangBambu