bambulab/BambuStudio #4186: Middle click paints instead of camera movement in paint mode

Namiiria, 2024-05-31:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

using mouse and keyboard

### Printer

A1 mini

### How to reproduce

Video of the issue with mouse inputs overlay: https://www.youtube.com/watch?v=xS_ber9661M

1. middle click outside of the model to move the camera and make the mouse go over the model
2. Without releasing the middle click, left click to paint
3. release everything
4. try to use the middle click to navigate again

To get unstuck: left click once outside the model, first time it doesn't rotate the camera, second time everything is back to normal

### Actual results

A stroke of paint is made on the object

### Expected results

The camera should move

### Project file & Debug log uploads

works on a brand new file, I cannot share this one right now.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-05-31:
Thank you for your discovery. We will fix it as soon as possible. Thank you

Haidiye00, 2024-06-03:
The problem has been resolved and is currently undergoing internal testing. Thank you for discovering the issue

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00