bambulab/BambuStudio #4187: Wrong File sent/printed when using Cloud Services

Crsarmv7l, 2024-05-31:
### Bambu Studio Version

1.9.1

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU i5-10500H, Ram 32GB

### Printer

A1 Mini

### How to reproduce

1. Generate a model in fusion, export it to .3mf
2. Open the model in Bambu Studio
3. Print
4. Close Bambu Studio
5. Modify the file in fusion
6. Save as new 3mf file name
7. Open that file name in Bambu studio and print
8. When opened in Bambu Studio you can see the correct file name
9. When it is sent to print, the printer will show and print the previous filename

### Actual results

The previous file prints and not the current one

### Expected results

The current file should be sent to the printer and printed

### Project file & Debug log uploads

N/a issue with the cloud service

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Crsarmv7l, 2024-05-31:
The first picture below you can see the file name is "Small Case" that was sliced and sent to the printer
[image]

This second picture you can see what the printer received/started printing, which was the previous file sent. (The same thing has happened multiple times but I just realized with this print). So I make changes to the model based on old no longer valid, old prints.
[image]

walterwongbbl, 2024-06-14:
@Crsarmv7l 
Hi. Does this model come from the maker world website