bambulab/BambuStudio #4192: Added part are no longer relative to the part they get attached to

Laserstrahl, 2024-06-01:
### Bambu Studio Version

v01.09.01.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1S

### How to reproduce

1. Add any STL to the plate, if you don't have, add an primitve.
2. Right-Click on the Part
3. Add Negative-Part --> Cube (or any other)
4. Click in the menu-band on the top on "move" or hit "M" key
5. now you can see the coordinates of the added negative part. Those parts are relative to the plate.

In the previouse version, when you follow this instrcutions, the coordinates where relative to the center of the object/stl to which you added the negative part (I call it root object). 

This is clearly a bug, since it makes it pretty complicated to position the additional (negative-)parts at the right spot in relation to the root object.

### Actual results

Coordinates of the added part are relative to the plate.

### Expected results

Coordinates of the added part are relative to the root object.

### Project file & Debug log uploads

Doesn't matter. you can simple reproduce it without any project files or STLs. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-06-03:
We have added a local coordinate system for objects in the internal version and are currently undergoing internal testing. Stay tuned.

Laserstrahl, 2024-06-03:
Alright. Actualy version v1.8.4.51 had this local coordinate system (coordinate system related to the root object) already. It worked pretty well. Was removed in v01.09.01.66

Haidiye00, 2024-06-04:
yes ,i kown

barneyvaughan, 2024-07-12:
Any update on this? Happy to support in testing if needed.

Haidiye00, 2024-07-15:
The next version of the move tool will be upgraded, please stay tuned and patiently wait.

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00