bambulab/BambuStudio #4194: 'Tree Hybrid' support ignores 'Top Z Distance' for some geometries, making support removal difficult

AJZein, 2024-06-02:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

Bambu Lab P1P

### How to reproduce

Download and slice the included .3mf file. Seems to only occur on curved overhangs.

### Actual results

The support interface has nearly no gap from the model making supports difficult to remove. Setting 'Top Z Distance' does not affect the gap at all (set here to 0.5mm)
[image]

### Expected results

The support interface should have the set gap from the model to facilitate easy removal. 'Tree Organic' seems to better handle this and respond to changes in 'Top Z Distance' as seen below
[image]

### Project file & Debug log uploads

[tree_hybrid_test.zip](https://github.com/user-attachments/files/15524878/tree_hybrid_test.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

AJZein, 2024-06-10:
After some more testing 'Tree Strong' and 'Tree Slim' are also affected by this issue

I've added a closer look below. Using Top Z Distance of 0.6mm and a layer height of 0.2mm, I would expect 3 layers of separation between interface and model. However there seems to be none in this example. For reference, normal grid supports show the expected 3 layer gap.

Layer 82, no interface on the left
[image]

Layer 83, interface constructed on the left
[image]

Layer 84, overhang already on interface. Next interface layer created
[image]

And the same repeats for the next layers
[image]
[image]

AJZein, 2024-06-11:
Updated to version 1.9.2.57, issue seems almost resolved with 'Independent support layer height' enabled. However when I disabled this to count the layers accurately the issue re-appeared 

Enabled
[image]

Disabled
[image]

tangketan, 2024-06-14:
Yes, synced support layer height still has issues... Thanks for reporting. I'll fix it soon.

ClawzSkunk, 2024-08-26:
> Yes, synced support layer height still has issues... Thanks for reporting. I'll fix it soon.

Can you advise when? I've had issues with supports for quite some time now

fahaaaxiki, 2024-10-08:
@ClawzSkunk Next Bambu Studio version will fix this problem.

ClawzSkunk, 2024-10-08:
Thankyou for the update. Is there an expected date for the next update?