bambulab/BambuStudio #4197: Since 1.9.X tree Support is not only on Heating plate

druckgott, 2024-06-02:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Intel(R) Core(TM) i5-4460  CPU @ 3.20GHz   3.20 GHz
8GB Ram

### Printer

P1P

### How to reproduce

Slize something which only should be on heatplate
[image]

Older Versions 1.8.x worked, if I slize it there it´s only on heatplate

### Actual results

It do not only put the tree on the heatplate, it´s also in the part

[image]

### Expected results

Only on Heatplate which worked on 1.8.x with exactly the same part

### Project file & Debug log uploads

I can´t upload a zip file because it´s a part which is not public

### Checklist of files to include

- [X] Log file
- [X] Project file

tangketan, 2024-07-19:
Yes, Tree slim could have this bug. It's an old bug. It has nothing to do with the "only on buildplate" option. Try "tree organic" instead.

druckgott, 2024-07-19:
I dont have this on Same parts in oder version. Its only on the. 1.9.xx version