bambulab/BambuStudio #4198: Text tool types text backwards (not mirrored, but letters in backwards order

DanDanBogBog, 2024-06-03:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma Version 14.5

### Additional system information

M1 Max chip

### Printer

P1S Selected

### How to reproduce

Unsure how to recreate, I suspect it's because the model had been mirrored. Object mirrored back along X-Axis did not fit it

### Actual results

[image]

Notice I had to type the text in reverse to get it to show up correctly

### Expected results

text works properly

### Project file & Debug log uploads

[Door Toppers - Text Bug.3mf.zip](https://github.com/user-attachments/files/15529488/Door.Toppers.-.Text.Bug.3mf.zip)

No log file required

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-06-12:
Hello, the issue is currently under review

Haidiye00, 2024-06-12:
I don't quite understand what you mean. Could you please describe your needs again.You can use some screenshots or GIFs, etc. to illustrate

RiverCrossDiorama, 2024-08-27:
Hello,
The problem still randomly appears on the latest 2 versions, including the 1.9.4.60
The text I try to type appears backwards when hovering over the object.
[image]

Haidiye00, 2024-08-28:
ok

tendeiro, 2024-10-02:
This issue is still happening in version 1.9.7.52, on Windows, when I have over >=12 plates.

Haidiye00, 2024-10-08:
Sorry, the resolved code has not been merged yet. Please wait

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

Haidiye00, 2026-09-22:
https://github.com/bambulab/BambuStudio/releases/tag/v02.08.04.57
Hello, please update to the latest version

super-structure, 2026-10-21:
I have a similar issue where the text used _in the actual model object_ gets reversed, sometimes at the letter / glyph value and sometimes the entire string.

MacOS v26.6.2 on Apple MacBook Pro M4
Bambu Studio v2.8.2.61 (latest public version as of this comment)

[image]

Haidiye00, 2026-10-22:
> 18\. 我也有类似的问题，实际_模型对象_中使用的文本会被反转，有时是字母/字形值，有时是整串。
>      苹果MacBook Pro M4上的MacOS v26.6.2 Bambu Studio v2.8.2.61（截至本文评论时为最新公开版本）
>      <img alt="图片" width="482" height="581" src="https://private-user-images.githubusercontent.com/98043/680642846-0b40c068-d655-4dbd-8eb3-0ad8d2a08b37.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3OTI2MzIwNjIsIm5iZiI6MTc5MjYzMTc2MiwicGF0aCI6Ii85ODA0My82ODA2NDI4NDYtMGI0MGMwNjgtZDY1NS00ZGJkLThlYjMtMGFkOGQyYTA4YjM3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNjEwMjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjYxMDIyVDAxMTYwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRlMWI4NDYxYjUwZmE2M2RmNThmNjQzNTEyMTgwN2M0OTYzMDdhZWY5M2E2ZTIzNDY4MWI1YmE1YjcwMDNmNjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JnJlc3BvbnNlLWNvbnRlbnQtdHlwZT1pbWFnZSUyRnBuZyJ9.KG4MqP4V6X-Hpoi3Jj5hChQAMtPU-D5zu0_onr4tfuc">

Hello, can you provide 3mf? Thank you

super-structure, 2026-10-22:
Here is the `.3mf` file with the reversing text issue: [https://www.icloud.com/iclouddrive/056N8TrGSpuwY7IkgkYR1wrKg#Schubox%5F5x5%5FHandles-Hinges-Metric](https://www.icloud.com/iclouddrive/056N8TrGSpuwY7IkgkYR1wrKg#Schubox%5F5x5%5FHandles-Hinges-Metric)

Haidiye00, 2026-10-23:
https://github.com/bambulab/BambuStudio/releases/tag/v02.08.04.57
Hi, I noticed you're not using the latest version（not Bambu Studio v2.8.2.61）. Could you please test version v02.08.04.57? Thanks, it seems fine on my end.

[image]

closed 2026-11-03 by Haidiye00