bambulab/BambuStudio #4201: [Regression]  Printer / Filament / Process dropdown menus do not display

jmickelin, 2024-06-03:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab github releases

### OS version

OS: Fedora 38

### Additional system information

Window manager: i3 4.22

### Printer

Any printer

### How to reproduce

Using Bambu_Studio_linux_fedora-v01.09.01.66.AppImage.

Example:

1. Go to the "Prepare" tab
2. Click the "Plate type" dropdown list

But it happens with _any_ dropdown lists in the entire program. Including the ones in dialog windows and the ones for "Slice plate" and "Print plate" in the top bar.

It _does not_ happen with the "menu bar" dropdowns (the hamburger "File" menu and the ∨-button immediately to the right of it).

In very rare cases I _have_ managed to get certain dropdown list to open properly, namely the infill and pattern dropdowns under the "Strength" tab. There does not seem to be any consistent setup to trigger it (just clicking around a bunch in the program, opening and closing some of the preference dialogs, moving the window to different workspaces on my desktop). Once it starts working, these dropdown lists (and only those!) seem to work until I restart the program. I have not managed to trigger this intermittently working behavior with any other dropdown list.

### Actual results

Dropdown list briefly flashes on screen for a fraction of a second, then disappears.

### Expected results

Dropdown list should stay open and allow you to select an option.

### Project file & Debug log uploads

[debug_Mon_Jun_03_14_24_06_34357.log.0.txt](https://github.com/user-attachments/files/15533995/debug_Mon_Jun_03_14_24_06_34357.log.0.txt)

### Checklist of files to include

- [X] Log file
- [ ] Project file

jmickelin, 2024-06-03:
This was originally reported downstream in OrcaSlicer: https://github.com/SoftFever/OrcaSlicer/issues/2101

Other users with different combinations of distro versions and desktop environments report the same issue.

The issue was _not_ present in BambuStudio 1.4.5, as per the findings I reported in this comment, hence I have marked this as a regression in the title of this bug report: https://github.com/SoftFever/OrcaSlicer/issues/2101#issuecomment-2041036989

DanBao-Bambu, 2024-06-13:
I am very sorry for the inconvenience caused to you, and the issue has been recorded.

smcgu, 2024-11-20:
Reverting the changes and compiling resolved this bug for me in 1.9.7.52.

smcgu, 2024-12-26:
@DrCaveDev and @DanBao-Bambu any word on when this will be fixed? I'm still running my patched 1.9.7.52.

gw0, 2024-12-30:
I hit the same problem with latest Bambu Studio 1.10.1.50 (Ubuntu 20.04 AppImage release, same with Flatpak release) in Qubes OS 4.2 in `debian-12-xfce` qube both using Xfce4 (also in `debian-11` qube using Gnome 3.38). These 3 dropdown menus are unusable, they show up for a split second then disappear. Same behavior for short or long click, right click, or scrolling. Because the UI is not designed for mouse-less usage it is also impossible to cycle through dropdown options with arrow keys. Xfce4 options such as disabling "Automatically give focus to newly created windows", "Automatically raise windows when they receive focus", "Raise window when clicking inside application window" "Activate focus stealing prevention", "Honor standard ICCCM focus hint", and "When a window raises itself: bring to current workspace" have no effect.

Confirmed that the issue is with `PopupWindow` (as @DrCaveDev determined) that is a thin wrapper around `wxPopupTransientWindow`. More specifically, there is a bug with the following condition that gets executed when you click on any dropdown menu and causes it to close. Everything works after commenting out this line and rebuilding.

```cpp
    if (!event.GetActive() && IsShown()) DismissAndNotify();
```
https://github.com/bambulab/BambuStudio/blob/master/src/slic3r/GUI/Widgets/PopupWindow.cpp#L35

gw0, 2025-01-06:
Maybe the proper solution for the alt+tab issue (reason for this regression) would be to use one of the GTK window type hints: https://docs.gtk.org/gdk3/enum.WindowTypeHint.html

closed 2025-01-14 by lanewei120