bambulab/BambuStudio #4202: Cannot find printer in LAN MODE if use another SSID Wifi

hechay, 2024-06-03:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab P1S

### How to reproduce

I get 2 Wifi SSID. One is 2.4GHz for the printer. One is 5GHz for my PC. My PC is normally connect to the network with cable plugged. Today, I got problem with the cable so I unpluged it. I use the 5Ghz Wifi Network for my computer. I cannot find the printer via BIND function in Bambu Studio. When I tried connect to 2.4GHz network, the printer can easily bind ( same with cable plugged ). Please understand the network is same ( subnet mark, router ip, etc... ). The different is only the name of network ( one for 2.4GHz and one for 5Ghz ). 

### Actual results

Cannot bind the printer when use 5Ghz network for computer although it was the same network only.

### Expected results

Hope you can have a function to bind the printer with IP address. It takes us more easily to connect the printer in LAN MODE. 

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/15536305/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

walterwongbbl, 2024-06-14:
Hi。 @hechay 
I'm very sorry for replying so late. We have received many such requests and we will add this feature in future versions. Thank you very much for your suggestion.

UltraBlackLinux, 2024-07-03:
running into a similar problem where I cannot even add the printer at all. It's not detected by bambu-studio, even though the printer is on the wifi, and I'm on ethernet. There really should be no communication problem and yet it's not connecting. Very weird.

The Add-by-IP suggestion would save us all a lot of hassle...

bandesz, 2024-12-22:
I have the same issue. If I connect my laptop to the printer's network, the printers do show up finally in Bambu Studio. After that I can connect back to the original Wifi network and connection will work with the printers until Bambu Studio is restarted.

L0RD4N, 2025-03-29:
Issue is still a thing, anyone found a permanent fix? Ty