bambulab/BambuStudio #4203: There is no Manual Flow dynamics Management Panel

Henlor, 2024-06-03:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 13.3.1

### Additional system information

Macbook air M1
Also tested on a Windows 10 (CoreI7-6700K) machine same result

Tested both app source Github and Official Website.

### Printer

Bambu Lab A1 Mini

### How to reproduce

Go to Calibration tab > Flow Dynamics

### Actual results

there is only one button at the bottom of the page.

[image]

### Expected results

https://wiki.bambulab.com/en/software/bambu-studio/release/release-note-1-9-1#h-8-support-manual-calibration-values-management

[image]

### Project file & Debug log uploads

There is no specific project open.

[no manual flow dynamics managment panel.zip](https://github.com/user-attachments/files/15536842/no.manual.flow.dynamics.managment.panel.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-06-04:
@Henlor  What is your firmware version? You can update to the latest firmware version and try again.

Henlor, 2024-06-04:
@zhimin-zeng-bambulab  Sorry I've forgot to mention this. I'm using the latest version of the A1 mini firmware 01.02.00.00

[image]

zhimin-zeng-bambulab, 2024-06-21:
The firmware version 1.2.0.0 does not yet support result management. You can wait for the next firmware version.

gaolegao-lx, 2024-06-24:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

Henlor, 2024-06-24:
@gaolegao-lx @zhimin-zeng-bambulab 
The issue was resolve by updating the printer with the latest firmware 1.03.00.00
Now it works perfectly 👌Thank you 🙏

closed 2024-06-24 by zhimin-zeng-bambulab