bambulab/BambuStudio #4205: Purging excessive amount at start of print to where it is clogging nozzle and prime wipe on A1 Single Material

kpdcompany, 2024-06-04:
### Bambu Studio Version

1.8.4.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu A1

### How to reproduce

Print TPU and see prime/purge at beginning of print.  Please see file.

### Actual results

Clogged prime wiper

### Expected results

Not clog, its priming an EXTREME amount

### Project file & Debug log uploads

[TPU PRIME ISSUE - PROJECT.zip](https://github.com/user-attachments/files/15541505/TPU.PRIME.ISSUE.-.PROJECT.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-06-11:
@kpdcompany  You are using TPU material, and then the printer gets clogged when it starts printing. Is that so？

zhimin-zeng-bambulab, 2024-06-11:
[image]

@kpdcompany 
The max volumetric speed is higher than the recommended value, which can easily lead to clogging. Could you try changing these parameters in the picture to the default values ​​before printing?

kpdcompany, 2024-06-11:
These did not change anything, this was doing it before I adjusted factory settings.  Also if these settings affect the priming then something is wrong and needs to be updated.  I dont understand why those factors would be involved in the prime.  Under no circumstance should the prime be this much filament, it just keeps going and going until it fills the wiper and jams it.  I dont understand why flow settings would make this such a huge purge when the amount of filament its primes should be the same no matter the flow rate or diameter.

zhimin-zeng-bambulab, 2024-06-21:
Do you mean that A1 is prone to clogging when printing TPU materials?
Could you describe your problem in more detail with picture?