bambulab/BambuStudio #4206: Long file names visible of models on SD card.

Hammer73, 2024-06-04:
**Is your feature request related to a problem? Please describe.**
I can't see the full file names of the items I send to the printer. I include printer type, nozzle size and filament as well as the name of the items. It quickly gets too long. 

**Describe the solution you'd like**
Be able to show a 'list' of files that allow long file names. 

**Describe alternatives you've considered**
Much abbreviation but it has caused errors.

**Additional context**
I

ethanzwolf, 2024-06-05:
Second this! Since our printers are used by multiple engineers, our work instruction designates the naming prefix to include nozzle size and material type at the start to prevent starting prints with the wrong nozzle. This makes it so most of the actual filename, including version number and date is not at all visible in the X1C file selection interface.  The fact that this is a rapid prototype device that doesn't allow you to see the full filename is totally unexpected and makes me want to return the X1C. **Please make the filename scroll when selecting a print file on the X1C** (just like on the A1). The A1 is a fraction of the cost of the X1C and even has this feature. 

Request image: Make filename scroll so entire filename is visible, not just part of it followed by "..." 
[image]

Reason this problem matters: Many iterations of the same part but you can't tell which one is which. 
[image]

GlennovitS, 2024-07-03:
Agree on this one - when prototyping new models this can be a little frustrating...
Following this for hopefully good news some day when fixed 👍🏼 

Would love it to be combined with this: [https://github.com/bambulab/BambuStudio/issues/4424](https://github.com/bambulab/BambuStudio/issues/4424) (feature request to customize how the default print names are generated)