bambulab/BambuStudio #4212: 亞太地區中文版無法使用不爾運算功能

1121stevenwang, 2024-06-04:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

[image]

### Printer

x1c   p1s   a1

### How to reproduce

亞太地區中文版無法使用不爾運算功能，下載軟體選亞太地區切英文後切中文，軟體變成中文後無法正常使用布爾運算，少了交集、差集

### Actual results

[image]

### Expected results

能正常使用

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

SaltWei, 2024-06-05:
看起来问题是中文字符没显示出来。
[image]

Haidiye00, 2024-06-05:
[image]
你好，想问一下，你的这个界面显示正常吗

1121stevenwang, 2024-06-05:
[image]
It displays fine and I can use the cutting function
[image]
Boolean functions cannot

I have tried it, it is also in the Asia-Pacific region. If you switch to English, you can use Bollinger.

1121stevenwang, 2024-06-05:
[image]
Just switch to English and you can use it.
The unavailability I mentioned earlier is not only that the button is not displayed, but also that I cannot click the button to use the intersection and difference functions.

Haidiye00, 2024-06-05:
https://www.dropbox.com/scl/fi/8my0l9zb7mzlabx9o3n38/Bambu_Studio_win_public-v01.09.02.54-20240605172311.exe?rlkey=bft2xdwg9lmos84a8vzuqdrcu&st=54gfzuis&dl=0
can you try this?thanks

tojo17, 2024-08-18:
[image]
这边也有完全一样的问题，切割功能正常

fahaaaxiki, 2024-10-08:
Hi，we can't reproduce this issue locally，can you help to update to the latest version [1.9.7.52 Public Official](https://github.com/bambulab/BambuStudio/releases/tag/v01.09.07.52) and check if the problem solved？

1121stevenwang, 2024-10-08:
還是無法使用，我電腦是使用繁體中文的，我猜是抓不到字?

fahaaaxiki, 2024-10-08:
> 還是無法使用，我電腦是使用繁體中文的，我猜是抓不到字?

我在我的电脑上也装了繁体字体，但是软件上有显示成功，很感谢您的帮忙，我再让我们的开发查看一下这个问题
[image]

tojo17, 2024-10-08:
我大概找到原因了，我的系统是日文的同样显示不出，但把非 unicode 程序的显示语言改成简体中文就可以显示了（中文新加坡一样可以，应该是编码问题），改回日文就无法显示。麻烦修复一下，感谢。

[image]
[image]

1121stevenwang, 2024-10-08:
如果可以，希望能有更詳細的中文說明或範例

tojo17, 2024-10-08:
@1121stevenwang 
简单贴了中文在上面，我不确定中文版系统这里具体是什么，但意思应该差不多，更改系统地区为简体应该就可以了，但这只是 workaround，因为改了这个会引起其他的程式乱码，还是需要拓竹解决这个问题。

[image]

1121stevenwang, 2024-10-08:
感謝 成功了

tojo17, 2024-10-16:
请问为啥 close 了，这个问题还是没有解决啊

Haidiye00, 2024-10-16:
Could you please test the 1.10beta version? Thank you~

tojo17, 2024-10-16:
@Haidiye00 Hi, I tried the latest [1.10.0 Public Beta](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.74), but no luck, under Japanese region set the text on buttons still could not be displayed.
[image]

fahaaaxiki, 2024-12-05:
> @Haidiye00 Hi, I tried the latest [1.10.0 Public Beta](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.74), but no luck, under Japanese region set the text on buttons still could not [...]

我们在1.10的正式版本[1.10.1 Public Release](https://github.com/bambulab/BambuStudio/releases/tag/v01.10.01.50)上再次尝试修复了此问题，可以使用这个版本帮忙确认一下是否正常了吗

tojo17, 2024-12-22:
@fahaaaxiki 
刚刚确认到这个版本里已经正常了，感谢跟进修复！

[image]

Haidiye00, 2024-12-23:
Hello, thank you for your reply. This is good news~

closed 2024-12-23 by Haidiye00